wlcrs/huawei_solar #4: Config flow could not be loaded

noobydp, 2021-10-30:
After installing the integration in HACS, restarting, then going to Add Integration > Huawei Solar, I get an error "config flow could not be loaded".

I dont get far enough to specify the IP address.

Please let me know if you need more info.

Here's a copy of the log

> Logger: aiohttp.server
> Source: custom_components/huawei_solar/const.py:80
> Integration: Huawei Solar
> First occurred: 5:17:51 PM (3 occurrences)
> Last logged: 5:21:01 PM
> 
> Error handling request
> Traceback (most recent call last):
>   File "/srv/homeassistant/lib/python3.8/site-packages/aiohttp/web_protocol.py", line 422, in _handle_request
>     resp = await self._request_handler(request)
>   File "/srv/homeassistant/lib/python3.8/site-packages/aiohttp/web_app.py", line 499, in _handle
>     resp = await handler(request)
>   File "/srv/homeassistant/lib/python3.8/site-packages/aiohttp/web_middlewares.py", line 119, in impl
>     return await handler(request)
>   File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/components/http/security_filter.py", line 60, in security_filter_middleware
>     return await handler(request)
>   File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/components/http/forwarded.py", line 98, in forwarded_middleware
>     return await handler(request)
>   File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/components/http/request_context.py", line 24, in request_context_middleware
>     return await handler(request)
>   File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/components/http/ban.py", line 78, in ban_middleware
>     return await handler(request)
>   File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/components/http/auth.py", line 144, in auth_middleware
>     return await handler(request)
>   File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/components/http/view.py", line 136, in handle
>     result = await result
>   File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/components/config/config_entries.py", line 130, in post
>     return await super().post(request)
>   File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/components/http/data_validator.py", line 62, in wrapper
>     result = await method(view, request, *args, **kwargs)
>   File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/helpers/data_entry_flow.py", line 72, in post
>     result = await self._flow_mgr.async_init(
>   File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/data_entry_flow.py", line 151, in async_init
>     flow, result = await task
>   File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/data_entry_flow.py", line 169, in _async_init
>     flow = await self.async_create_flow(handler, context=context, data=data)
>   File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/config_entries.py", line 709, in async_create_flow
>     integration.get_platform("config_flow")
>   File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/loader.py", line 532, in get_platform
>     cache[full_name] = self._import_platform(platform_name)
>   File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/loader.py", line 537, in _import_platform
>     return importlib.import_module(f"{self.pkg_path}.{platform_name}")
>   File "/usr/local/lib/python3.8/importlib/__init__.py", line 127, in import_module
>     return _bootstrap._gcd_import(name[level:], package, level)
>   File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
>   File "<frozen importlib._bootstrap>", line 991, in _find_and_load
>   File "<frozen importlib._bootstrap>", line 961, in _find_and_load_unlocked
>   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
>   File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
>   File "<frozen importlib._bootstrap>", line 991, in _find_and_load
>   File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
>   File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
>   File "<frozen importlib._bootstrap_external>", line 843, in exec_module
>   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
>   File "/home/homeassistant/.homeassistant/custom_components/huawei_solar/__init__.py", line 8, in <module>
>     from .const import DOMAIN, CONF_SLAVE, DATA_MODBUS_CLIENT
>   File "/home/homeassistant/.homeassistant/custom_components/huawei_solar/const.py", line 80, in <module>
>     SENSOR_TYPES: tuple[HuaweiSolarSensorEntityDescription] = (
> TypeError: 'type' object is not subscriptable

[image]

wlcrs, 2021-10-31:
Hi, can you please provide me with some info on your HA instance? Which version are you using? I suspect there is a mismatch in Python version, as this plugin was only tested against the latest HA version, which is using Python 3.9 instead of the 3.8 I see mentioned in you error logs

noobydp, 2021-11-05:
Hey, you’re right I’m running python 3.8. I installed HA manually on an RPI 4B with Raspbian Lite which runs python 3.7 by default. 

I’ll update python and try again.

closed 2021-11-05 by noobydp