wlcrs/huawei_solar #7: Huawei Solar not avaible from the list "Add Integration"_ wlcrs / huawei_solar

pascalppp, 2021-11-20:
Hello,

I have done all steps:
1. Install integration with HACS, or copy the contents of this repository into the custom_components/huawei_solar directory
2. Restart HA
3. Go to Configuration -> Integrations and click the + Add Integration button
And Huawei Solar ist not avaible to be selected. 
I have huawei_solar folder with all components, in HACS this Integration Hawei Solar is visible, but nothing in the Configuration/Integration - so I can not edit Energy card.
Probably this is sth simple but I can not find solution - Thank you for your help in adv!

wlcrs, 2021-11-20:
Can you post a listing of your config-folder and the logs from HA after
restarting it? I have too little information to work with to give any
concrete advice.

Emilv2, 2021-11-20:
Huawei solar does not use config flow but yaml configuration, you will have to add the configuration to your `configuration.yaml` file. Using config flow is on the todo list.

pascalppp, 2021-11-22:
Sorry for my late reply,

All informations below
[image]
[image]
[image]
[home-assistant.log](https://github.com/wlcrs/huawei_solar/files/7582758/home-assistant.log)
[image]
[image]

Logs details:
Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:164
Integration: Sensor (documentation, issues)
First occurred: 16:17:04 (1 occurrences)
Last logged: 16:17:04

The huawei_solar platform for the sensor integration does not support platform setup. Please remove it from your config.

Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/connection.py:89
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 16:17:01 (2 occurrences)
Last logged: 16:17:01

[547743223280] Received invalid command: energy/info

wlcrs, 2021-11-23:
From the logs/screenshots I can see that the huawei_solar custom component is being loaded correctly. The error is correct and expected: the component from this repo must be configured via the "Integrations" menu, as described in the README. 

It's the integration of Emilv2 that uses the "platform setup" via `configuration.yaml`, so those lines should be removed. (I guess he reacted in this issue thinking it was about his own integration? )

The other error seems to be caused by having an invalid entity to record your energy usage: https://github.com/home-assistant/core/issues/54060

Which version of HA are you using? Can you increase the logging of your instance by adding the lines below to `configuration.yaml` and post the full (anonymized) home-assistant.log somewhere for me to inspect?

```
logger:
  default: debug
  logs:
    custom_components.huawei_solar: trace
```

Emilv2, 2021-11-23:
@wlcrs you are right :) disregard what I said.

pascalppp, 2021-11-23:
I removed those lines from configuration.yaml, restarted HA (version core-2021.11.4) and "Huawei Solar" appeared in the Integration to choose (before I left those lines from Emilv2 Integration (which I tried as well before) in configuration.yaml, because I couldn't choose "Huawei Solar" from the Integration and I didn't have place to write IP of Inverter - I thought, maybe It will be work this way too, but It didn't :)
So I configured Energy Card and I'm waiting for tomorrow, when Sun appears :)

One problem I have is the date - Energy Card shows me always the previous date as default (so today 22.11- see screenshot)   and I can not choose "today" - but this is on the website, on the mobile application on iOS I have correct dat
[image]
e

wlcrs, 2021-11-23:
Glad to hear that the issue with this integration was resolved.

closed 2021-11-23 by wlcrs

pascalppp, 2021-11-23:
Thank You for your help! :)