wlcrs/huawei_solar #8: Huawei Solar not avaible from the list "Add Integration"_ wlcrs / huawei_solar #7

Marchellos, 2021-12-01:
The [same problem](https://github.com/wlcrs/huawei_solar/issues/7#issue-1059149159) with "Huawei Solar is not available in the list"

The component appears to be loaded from the log.

2021-12-01 15:52:17 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant

2021-12-01 15:52:17 WARNING (SyncWorker_1) [homeassistant.loader] **We found a custom integration huawei_solar** which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant

2021-12-01 15:52:17 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration sonoff which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant

Furthermore, there were no messages in the log.
But the component Huawei Solar not avaible from the list "Add Integration".
Nothing has changed in the /config/configuration.yaml. That's why I didn't include this one.

## System Health

version | core-2021.11.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.75
arch | x86_64
timezone | Europe/Amsterdam

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4892
Installed Version | 1.18.0
Stage | running
Available Repositories | 975
Installed Repositories | 3

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 6.6
-- | --
update_channel | stable
supervisor_version | supervisor-2021.10.8
docker_version | 20.10.8
disk_total | 30.8 GB
disk_used | 2.6 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | SSH & Web Terminal (9.1.1), File editor (5.3.3), Filebrowser (2.19.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 0
views | 1
mode | storage

</details>

wlcrs, 2021-12-01:
The warning is expected and normal. It does indicate that the component was loaded. Without any additional warning/error messages there is very little I can do or suggest.

Can you please increase the logging level of your HA by adding the following lines to `configuration.yml`:

```
logger:
  default: debug
  logs:
    custom_components.huawei_solar: trace
```

Hopefully this yields somewhat more information.

Marchellos, 2021-12-01:
Trace is not a valid debug level. I chose "ïnfo"

Failed to connect: [Errno 111] Connect call failed ('192.168.0.123', 502)
18:09:41 – (WAARSCHUWING) /usr/local/lib/python3.9/site-packages/pymodbus/client/asynchronous/async_io/__init__.py - bericht kwam voor het eerst om 18:09:28 en verschijnt 8 keer

Setup of sensor platform huawei_solar is taking over 10 seconds.

18:09:38 – (WAARSCHUWING) Sensor could not connect to Huawei inverter: failed to connect to device, is the host correct?
18:09:30 – (FOUT) huawei_solar (aangepaste integratie) failed to connect to device, is the host correct?
18:09:30 – (FOUT) /usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py

wlcrs, 2021-12-01:
I assume that you were able to find the integration in the list, that you did input 192.168.0.123 as host in the configuration screen?

In my case, the inverter (installed in October of this year) initially did not expose its modbus interface on the network to which it connected. Only after upgrading to the latest firmware this started to work. Before the upgrade, I had to connect to the WiFi broadcasted by the inverter itself, and only there the modbus interface was available on 192.168.200.1 .

closed 2021-12-07 by wlcrs