wlcrs/huawei_solar #9: Huawei Solar not showing in integrations list

Shortblade, 2021-12-21:
Hi,

after adding Huawei Solar to HA via HACS, it's not showing in the integration list.
I tried adding Huawei Solar in HACS by copying the huawei_solar folder to the custom_components folder.
That didn't make Huawei Solar visible in HACS so i removed the folder again and then i added Huawei Solar to HACS true Custom Repositories as an integration.
After that i restarted HA and Huawei Solar was now showing in HACS with a download option.
So i downloaded the Huawei Solar integration and did a restart as asked by HA.
Now that Huawei Solar is installed in HACS, i don't see it showing up in the integrations list in the configuration section.
Configuration -> Devices & Services -> + Add Integration - Huawei Solar is not in the list.
I tried multiple restarts.

Shortblade, 2021-12-21:
Hi,

can you tell me what i'm looking for in the log file?
There are 9604 lines so i'm not gonna paste that here.
This is the short log:
2021-12-21 16:31:19 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration fusion_solar_kiosk which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-12-21 16:31:19 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-12-21 16:31:19 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration tuya_v2 which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-12-21 16:31:19 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration huawei_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-12-21 16:31:22 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [548451958544] Received invalid command: energy/info
2021-12-21 16:31:22 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [548451958544] Received invalid command: energy/info
2021-12-21 16:31:22 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [548451958544] Received invalid command: energy/info
2021-12-21 16:31:22 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [548451958544] Received invalid command: energy/info
2021-12-21 16:31:22 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [548451958544] Received invalid command: energy/info
2021-12-21 16:31:22 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [548451958544] Received invalid command: energy/info
2021-12-21 16:31:31 ERROR (MainThread) [homeassistant.components.forecast_solar] Unexpected error fetching forecast_solar data: Rate limit for API calls reached. (error None)

wlcrs, 2021-12-21:
Searching for huawei_solar should give you the interesting lines for me to
have a look at. Also any errors that are possibly related should be
included.

closed 2021-12-29 by wlcrs