wlcrs/huawei_solar #11: [Bug]: Integration does not show up

tgrosz, 2021-12-29:
### System Health details

## System Health

version | core-2021.12.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.63-v8
arch | aarch64
timezone | Europe/Budapest

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4776
Installed Version | 1.18.0
Stage | running
Available Repositories | 933
Installed Repositories | 7

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.0
-- | --
update_channel | stable
supervisor_version | supervisor-2021.12.2
docker_version | 20.10.9
disk_total | 28.6 GB
disk_used | 6.4 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Check Home Assistant configuration (3.9.0), Grocy (0.11.0), Portainer (2.0.0), File editor (5.3.3), Log Viewer (0.12.1), FTP (4.4.0), ESPHome (2021.12.1), Let's Encrypt (4.12.0), NGINX Home Assistant SSL proxy (3.1.0), SSH & Web Terminal (10.0.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 2
-- | --
resources | 3
views | 5
mode | storage

</details>

### Describe the issue

Hi, I do not know what I do wrong. I added the repo to HACS, installed it. The folder with the files is under custom_components. HA is restarted. Nothing Huawei related is in the logs.
Still the integration is not in the list of available integrations.

### Reproduction steps

1. Try to add the integration
2.
3.
...

### Relevant debug logs

```text
nothing unfortunately
```

wlcrs, 2021-12-29:
I did some (unrelated) updates to this integration today, and tested it on a fresh HA installation to verify its proper working.

Adding the integration on HACS involves 2 steps:

1. Using the top-right menu to find 'Custom repositories' and adding the repository:
[image]

2. Press the blue `+ Explore and download repositories` button in the bottom-right and select the `Huawei Solar` integration which has now become available in that list.

The Huawei Solar integration must now be visible in the HACS screen:
[image]

3. After restarting HA, the red banner disappears:

[image]

4. Your HA logs should at the least mention the following lines:

```
2021-12-29 16:04:28 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-12-29 16:04:28 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration huawei_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
```

5. The custom integration is now available in the `+ Add integration` list:
[image]

Please let me know if this explanation resolved your problem. Also mention what error you made and how you resolved it, so that others can learn.

tgrosz, 2021-12-29:
Hi, thank you - now HACS said there’s a new version, I updated. Then the integration appeared in the list, I could install. It is not working yet (could not connect, and modbus errors in log). I guess they are because I have to upgrade the firmware of the dongle and the unit also - waiting for installer guys to do it.

closed 2021-12-30 by wlcrs