wlcrs/huawei_solar #12: config flow could not be loaded

Ryuu125, 2022-01-07:
After installing Huawei Solar i cannot get past the config flow could not be loaded warning message, I have checked the closed issue in #4 and i have python version 3.9.7 installing on a Hyper V system using HA 2021.12.8
[image]

The only error that does come up is this
Logger: homeassistant.config_entries
Source: config_entries.py:720
First occurred: 12:24:27 PM (1 occurrences)
Last logged: 12:24:27 PM

Error occurred loading configuration flow for integration huawei_solar: cannot import name 'AsyncHuaweiSolar' from 'huawei_solar' (/usr/local/lib/python3.9/site-packages/huawei_solar/__init__.py)

wlcrs, 2022-01-07:
Hi, this error indicates that the dependency for package `huawei-solar==1.2.0` specified in `manifest.json` (here: https://github.com/wlcrs/huawei_solar/blob/main/manifest.json#L7) was not installed correctly. Because of that, the class `AsyncHuaweiSolar` is missing. I checked pypi, and the package `huawei-solar` is still published, so the issue must be on your side.

Can you try a full reinstall (by removing the folder from custom_components, restarting HA, adding the integration again, restarting again) to check if you can find any relevant information on why the installation of that dependency failed? You should find some indications in the logs after restarting HA when you added the custom_component again.

Ryuu125, 2022-01-07:
Hi there
Yeah doing a full reinstall like removing it from the components section completely and reinstalling it again has not changed anything. The logs arent completely showing anything for why it failed besides the same error log.

wlcrs, 2022-01-07:
I tested the latest release in a clean Home Assistant installation less than two weeks ago. If your Home Assistant installation doesn't properly install dependencies defined in the `manifest.json` file, then I'm afraid you'll need to investigate that further.

I'm closing this issue as it is an issue with your HA installation, and not caused by an error in this integration.

closed 2022-01-07 by wlcrs

gieljnssns, 2022-01-21:
I have a similar error
```
Logger: homeassistant.config_entries
Source: config_entries.py:720
First occurred: 14:33:11 (1 occurrences)
Last logged: 14:33:11

Error occurred loading configuration flow for integration huawei_solar: cannot import name 'is_installed' from 'pymodbus.compat' (/usr/local/lib/python3.9/site-packages/pymodbus/compat.py)
```

[image]

wlcrs, 2022-01-21:
Please open your own issue, and fill in the template to provide me with the necessary information about your installation.