wlcrs/huawei_solar #13: huawei_solar.huawei_solar.ReadException: could not read register value, is an other device already connected?

Marchellos, 2022-01-07:
### System Health details

## System Health

version | core-2021.12.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.88
arch | x86_64
timezone | Europe/Amsterdam

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4982
Installed Version | 1.18.0
Stage | running
Available Repositories | 932
Installed Repositories | 3

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.1
-- | --
update_channel | stable
supervisor_version | supervisor-2021.12.2
docker_version | 20.10.9
disk_total | 30.8 GB
disk_used | 4.8 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.3.3), Filebrowser (2.20.1), Terminal & SSH (9.2.2), MariaDB (2.4.0), phpMyAdmin (0.6.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 0
views | 2
mode | storage

</details>

### Describe the issue

[... 89 lines cut ...]
File "/config/custom_components/huawei_solar/config_flow.py", line 77, in async_step_user
info = await validate_input(self.hass, user_input)
File "/config/custom_components/huawei_solar/config_flow.py", line 47, in validate_input
model_name = (await inverter.get(ATTR_MODEL_NAME)).value
File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 394, in get
response = await self.read_register(await self.client, reg.register, reg.length)
File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 438, in read_register
raise ReadException(message)
huawei_solar.huawei_solar.ReadException: could not read register value, is an other device already connected?
2022-01-07 14:26:16 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 502, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 502)
2022-01-07 14:26:16 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 502, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 502)
2022-01-07 14:26:16 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 502, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 502)
2022-01-07 14:26:17 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 502, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 502)
2022-01-07 14:26:19 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 502, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 502)
2022-01-07 14:26:22 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 502, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 502)
2022-01-07 14:26:28 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 502, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 502)
2022-01-07 14:26:41 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 502, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 502)
2022-01-07 14:27:07 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 502, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 502)
2022-01-07 14:27:58 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 502, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 502)
2022-01-07 14:29:40 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 502, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 502)
2022-01-07 14:33:05 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 502, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 502)
2022-01-07 14:38:05 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 502, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 502)
2022-01-07 14:43:05 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 502, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 502)
2022-01-07 14:48:05 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 502, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 502)
```

wlcrs, 2022-01-07:
As I already said to you in #3: The inverter closes the connection instead of answering the query after issuing a read_register (0x03) command, so having an open 502 port to connect to is clearly not enough.

The device will always close the previous connection when another connection is made to that 502 modbus port. So make sure that nothing else is trying to connect to the inverter via Modbus. This can also cause the behavior that you're seeing.

To debug, you should verify that you can succesfully retrieve a value via Modbus from the inverter.

I've only worked with Modbus directly via Python. A minimal example to debug:

```
from pymodbus.client.sync import ModbusTcpClient as ModbusClient
client = ModbusClient('<ip>', port=502)
client.connect()
client.read_holding_register(address=30000, count=15 ,unit=0)
```

If you want to use a GUI to query, you'll have to search for yourself (maybe http://www.minaandrawos.com/tcp-modbus-examiner-tool/ works for you?)

Marchellos, 2022-01-07:
Hello Thijs,

Thank you very much for your help.
I am sure there are no other connections.
My inverter hangs in a barn in the countryside.
The inverter can only connect to the router.
Huawei recommended SPC126.
By default, the modbus is turned off.
But to enable the modbus I need an android phone.
And I only have an iPhone cellphone.
I just got this response from the helpdesk:

I escalate this issue and R&D confirmed that this month they will enable
the option for Sdongle MODBUS TCP through Fusion Solar Web!

Until then we cannot do anything.

 We are sorry for the inconveniences!

 I wish you a nice evenin

So I will have to be patient until the update comes.
In the meantime I will study modbus protocol to understand this.

Thanks again for your help and have a nice weekend..

wlcrs, 2022-01-07:
It may be possible that you'll need to contact your installer to enable it via that way. My account certainly doesn't have enough access rights for that kind of changes to the installation.

closed 2022-01-07 by wlcrs