wlcrs/huawei_solar #14: Wrong Data on Energy Dashboard

jsagu, 2022-01-10:
Hi Guys

First many thanks for the amazing job. Im new to HA and just taken my first steps on it. 

I followed the readme file and I got my inversor connected and providing data but what I get looks really strange. 

[image]

[image]

[image]

I was starting to play around with the data but before wanted to ensure that I did not put anything wrong.  BTW Inversor is a Huawei SUN2000 5KTL

wlcrs, 2022-01-10:
I've identified the cause this weekend and was trialing the fix.

With the previous-to-last update, it was possible that HA was bombarding the inverter with update requests, causing it to return 0 instead of the proper value. This caused HA to think that the value had reset, leading to "flickering" between 0 and the proper daily value. HA sums the values between each reset, leading to the unrealistic high values.

I've extracted the fix from all my diagnostic code that I created to find the issue and pushed it to the repo.  Untested for the moment, but hopefully it works for you?

[image]

jsagu, 2022-01-10:
Hi Thijs

Thanks for the answer I just added the new sensor into my HA and works perfectly for the moment. 

Python is not part of my expertise but if I can help in any way please ping me. 

B&R
J.

closed 2022-01-12 by wlcrs