wlcrs/huawei_solar #15: Add battery "Maximum Discharging Power" as a configurable NumberEntity

wlcrs, 2022-01-12:
I’ve been working on exposing some parameters as configurable entities in Home Assistant. My primary use case would be to make it possible to set the “Maximum Discharge Power” for my LUNA2000 battery from Home Assistant. This would allow me to adjust the power output of the battery dynamically based on other inputs.

Example use case: in Belgium, you are taxed on peak usage, starting when you draw more than 2.5kW from the grid for a certain time. I would like to create an automation that sets the “Maximum Discharging Power” for my battery to “Current Grid Power Usage (in W) - 2500W”.

However, while the register is mentioned as being read/writeable in the Modbus Interface Definitions document, I’m receiving a “Permission denied” error (0x80) when trying to write to it.

[image]

Support is letting me know that the function is not available at the moment :frowning: :

[image]

For everyone who would also be interested in using this function: Please send a mail to Huawei support to ask for this functionality! The more requests are made, the more hope that the function will ever be enabled.

An example mail:

> Dear,
> 
> I want to use the Modbus-interface on my inverter to dynamically set the maximum discharging power of my LUNA2000 battery.
> The Modbus Interface Definitions state that register 47077 is RW. However, when I try to write a value to this register, I get the error code 0x80, which means “No permission”?
>
> My inverter SN: HVxxx> 
> Battery SN: BTyyy

inigoml, 2022-01-25:
Hi @wlcrs 

It's a very interesting feature. Not only discharge power, but also charge/discharge mode (maximum autoconsumption on sunny days, TOU in cloudy days or winter for charging in low price hours), definable TOU charge&discharge hours or battery capacity grid cutoff percentage based on energy generation forecast in order to establish a low charge level for sunny days or higher charge level for cloudy days.

Will write to EU Support also.

wlcrs, 2022-01-25:
> Done!
>
> Good night,
>
> I'm trying to use the modbus-interface to dynamically set several
> parameters of my LUNA2000 battery. In particular, I want to change charge
> level from the grid for TOU mode based in solar production forecast. For
> example, for sunny days I would start at first hour in the morning with 20%
> because my battery would charge from solar panels during the day, while for
> cloudy day I would start at first hour with 80% or 100% due to a fair
> amount of power is going to be generated. What I looking for is to avoid
> peak hours in terms of energy price from grid. Usually in Spain nights are
> really cheap and so I pretend to load battery at the night based in what
> I'm going to produce during day.
> The Modbus Interface Definitions state that register 47077 is RW. However,
> when I try to write a value to this register, I get the error code 0x80,
> which means “No permission”?
> My inverter SN: XXXXXXXXXXXXXXX
> SW version: V200R001C00SPC117
> Model: SUN2000-4KTL-L1
> Battery SN: XXXXXXXXXXXXXXXXX
> SW version: V100R002C00SPC106
> Model: LUNA2000-10KWH
>
> Thanks in advance.

Hi,

An update from my side. I pressed Huawei Europe for some more information,
and they informed me that an SDongle is needed in order to be able to write
to the registers to my SUN2000-3K-L1. The modbus interface which is exposed
by my inverter directly is not authorized apparently.

I'm now in contact with my installer to figure out the next steps. But it
takes some time.

I'm currently finishing a big rewrite/update from Emilv2's library that
does the actual modbus communication. Once that is done, I'll submit an
official integration for HA (based on the work done in this repo).
Initially this integration will only read from the inverter, but
write-support is still on the roadmap after that.

inigoml, 2022-01-26:
Since you can alter inverter behavior from SUN2000 application, for sure we
can alter inverter parameters from SUN2000-XXXXX network. Not sure if
SUN2000 app uses modbus protocol or other Huawei custom one for modify
parameters but this way for sure exists.
Another way would be using REST API from Fusionsolar web. But you need an
account with installer privileges at website.

inigoml, 2022-01-26:
Response from Huawei support:
[image]

inigoml, 2022-01-26:
They also sent me this document:
[image]

inigoml, 2022-01-26:
@wlcrs, do you have some code snippet in order to test with other modbus commands?
I'm trying to modify this:

[image]

And also this:
[image]

inigoml, 2022-01-27:
@wlcrs, do you have some code snippet in order to test with other modbus commands?
I'm trying to modify this:

[image]

And also this:

wlcrs, 2022-01-27:
I have a branch with the poc for writing to registers. Have a look at this:
https://gitlab.com/wlcrs/huawei-solar/-/blob/write-support/temp.py

wlcrs, 2022-01-30:
I’ve worked out how to do the login on the inverter in the same way the FusionSolar app does. This enables write-support for some configuration options, all related to having a battery (maximum charge power, maximum discharge power, charging from the grid, …).

[image]

You can test it by using the code from the `login-and-write` branch. Note that it requires the installer account to have the default password 00000a for now.

I haven’t been able to test this new functionality for long-term stability yet. Please report any unexpected behavior to me.

inigoml, 2022-01-30:
Wow... you are my hero!

inigoml, 2022-01-30:
Working but... also not writable. 
[image]

wlcrs, 2022-01-30:
That will be a mistake from my end, not an error returned by the inverter.
I'll fix it tomorrow.

In the meantime: I also implemented a few services which allow you to start
a forcible charge/discharge for a certain duration of time, or until a
specific SoC is reached. Check them in the services tab of the developers
tools

inigoml, 2022-01-30:
I'm moving right now from emilv2 huawei_solar to your branch. I had to remove all fixes and workarounds in order to make Energy Stats working. Yours is pretty clean!

inigoml, 2022-01-30:
With this branch you could eventually eliminate the need of install SUN2000 commissioning app!

[image]

wlcrs, 2022-01-30:
Just tested those services again myself, but it seems like I have broken it
just before I finished working on them. Will look into it next week.

wlcrs, 2022-01-31:
I've fixed a few register-settings, and also added a "stop forcible charge" service. Can you do a 'git pull' and try again?

I've tested the services, and they seem to be working well now. 

An observation: the forcible charge commands will not charge from grid if the solar panels are producing anything at all. Not surprising that the inverter can't do both at once. Instead, all production will be diverted into the battery until the target is reached.

inigoml, 2022-01-31:
It appears is working right now (no error). Will check later via SUN2000 to ensure values are also set up at inverter.
This is a really nice feature!

JacoboSanchez, 2022-02-15:
I am posting here as it is somehow related with the topic. I am testing the write feature to figure out how to automate a battery load at night with lower prices and I am getting sometimes permission denied errors. My guess is that previously authentication is somehow lost and the inverter requires to authenticate again. Did anyone experienced this? 

FYI: I am still using the network interface on SP109 firmware (not SDongle and not connected to the built-in access point)

inigoml, 2022-02-15:
I've tested this feature with these firmware versions:

SPC114 -> Modbus connection to inverter IP in my wifi network (port 502), and modbus connection to inverter IP in inverter wifi network (SUN2000-XXXXXXX), port 502.
SPC117 -> Modbus connection to inverter IP in inverter wifi network (SUN2000-XXXXXXX), port 6607
SPC119 -> Modbus connection to inverter IP in inverter wifi network (SUN2000-XXXXXXX)., port 6607

Feel free to upgrade to SPC114 but not go further. In general, SPC114 is the best one due to no wifi network issues and modbus opened in regular wifi (SPC117 and SPC119 has "once a day" disconnection in my wifi network and I have to enter via FusionSolar commissioning app in order to reconnect wifi).

wlcrs, 2022-02-15:
Can you increase your logging (cfr. bottom of the FAQ in the README), wait till the issue occurs, and post (a link to) your home-assistant.log in a new issue?

As I do frequent restarts of my own HA-instance (because I need to test small fixes here and there), I haven't been able to test long-running sessions myself.

I want to know if the underlying logic is sound before I add retry-mechanisms to login again.

inigoml, 2022-02-15:
> I want to know if the underlying logic is sound before I add retry-mechanisms to login again.

Regarding retry, due to frequent loss of connectivity with inverter (at least for those models with internal wifi, not sure with FE dongle), sometimes automations do not work due to reconnection is read-only. I don't know why. Only way to fix this is restarting HomeAssistant.

JacoboSanchez, 2022-02-15:
> Can you increase your logging (cfr. bottom of the FAQ in the README), wait till the issue occurs, and post (a link to) your home-assistant.log in a new issue?
>
> As I do frequent restarts of my own [...]

I have activated the log (pymodbus included) so I will create another Issue when reproduced, thanks.

JacoboSanchez, 2022-02-15:
Well it refuses to fail now... maybe I must wait until tomorrow. In the future I plan to install a modbus  rtu-tcp bridge to be able to connect without using Huawei's modbus tcp implementation. This way I should have write permissions without login... At that point I will test if that works.
I already have one inverter using the bridge (Elfin EW11) but it has no battery so I have no write options to test

wlcrs, 2022-02-15:
The integration should already support this scenario. When you connect over your rtu-tcp bridge, and check the "Advanced: enable parameter configuration" checkbox, it should skip the screen in which it asks for the installer username/password. 

I would greatly appreciate it if you let me know if that works as expected or not.

You could already test it with your current setup. Checking that checkbox will just try to write the timezone-register. It should not produce any errors. It will not produce any configurable entities of course.

JacoboSanchez, 2022-02-15:
[image]
[image]
And... maybe this is the proof?
[image]

wlcrs, 2022-02-15:
The pymodbus logs are from a heartbeat of your other inverter. But it's very good news that using your rtu-tcp bridge worked so smoothly!

Are your inverters not connected to each other via the RS485_2 interface? You should be able to read from both of them by connecting to the "master" inverter and mentioning all slave ids (comma separated) in the first config screen.

denislooby, 2022-02-22:
It's interesting that the docs say the range for the Grid Charge Cutoff SOC is 30-70 but using fusion solar and the installer account directly connected I can use higher cutoff than that.
I wonder if it varies by inverter or battery, LG vs Luna, it's a Luna in my case.

wlcrs, 2022-02-22:
@denislooby . Yeah, I first implemented it as per the modbus register definitions, but in #25 this was changed to reflect the reality of what FusionSolar allows.

I'm closing this issue as this feature has been adequately tested by various people now

closed 2022-02-22 by wlcrs