wlcrs/huawei_solar #17: [Bug]: Error occurred loading configuration flow for integration huawei_solar

gieljnssns, 2022-01-21:
### System Health details

## System Health

version | core-2021.12.10
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.88
arch | x86_64
timezone | Europe/Brussels

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 5000
Installed Version | 1.19.3
Stage | running
Available Repositories | 1017
Downloaded Repositories | 36

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.1
-- | --
update_channel | beta
supervisor_version | supervisor-2021.12.2
docker_version | 20.10.9
disk_total | 457.7 GB
disk_used | 113.9 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Samba share (9.5.1), AdGuard Home (4.4.2), AppDaemon 4 (0.8.1), Caddy (1.6), Check Home Assistant configuration (3.9.0), Duck DNS (1.14.0), ESPHome (2022.1.1), Mosquitto broker (6.0.1), Studio Code Server (4.1.0), WireGuard (0.6.0), MariaDB (2.4.0), Samba Backup (5.0.0), InfluxDB (4.3.0), Grafana (7.4.1), Frigate NVR (2.3), SSH & Web Terminal (10.0.2), Glances (0.14.1), phpMyAdmin (0.7.0), Mealie (0.5.4), Arpspoof (1.0.0), Guacamole (1.3.0), Caddy 2 (1.0.9), Firefly iii (5.6.10-7)

</details>

<details><summary>Lovelace</summary>

dashboards | 3
-- | --
resources | 17
views | 18
mode | storage

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Describe the issue

Error occurred loading configuration flow for integration huawei_solar

### Reproduction steps

1. Try to setup Huawei_solar
2.
3.
...

### Relevant debug logs

```text
Logger: homeassistant.config_entries
Source: config_entries.py:720
First occurred: 14:33:11 (1 occurrences)
Last logged: 14:33:11

Error occurred loading configuration flow for integration huawei_solar: cannot import name 'is_installed' from 'pymodbus.compat' (/usr/local/lib/python3.9/site-packages/pymodbus/compat.py)
```

gieljnssns, 2022-01-21:
[image]

wlcrs, 2022-01-21:
What is the output of `pip freeze` in the HA docker container ? Do you have any other integrations that also use modbus? 

How did you add the plugin to your HA? I recently tested the integration on a fresh install, and wrote down the exact steps I followed here: https://github.com/wlcrs/huawei_solar/issues/11#issuecomment-1002671977

gieljnssns, 2022-01-22:
pip freeze
```
atomicwrites==1.4.0
attrs==21.2.0
audio-metadata==0.11.1
auroranoaa==0.0.2
aurorapy==0.2.6
Authlib==0.15.5
awesomeversion==21.11.0
axis==44
azure-common==1.1.27
azure-core==1.21.1
azure-eventhub==5.5.0
azure-servicebus==0.50.3
Babel==2.9.1
backoff==1.11.1
baidu-aip==1.6.6.0
base36==0.1.1
batinfo==0.4.2
bcrypt==3.1.7
beautifulsoup4==4.10.0
bellows==0.29.0
bidict==0.21.4
bimmer-connected==0.8.10
bitarray==2.3.5
bitstring==3.1.9
bitstruct==8.12.1
bizkaibus==0.1.1
bleak==0.14.1
blebox-uniapi==1.3.3
blinkpy==0.17.0
BlinkStick==1.2.0
blockchain==1.4.4
blurhash==1.1.4
bond-api==0.1.15
boschshcpy==0.2.19
boto3==1.20.38
botocore==1.23.38
bravia-tv==1.0.11
broadlink==0.18.0
brother==1.1.0
brotlipy==0.7.0
brottsplatskartan==0.0.1
brunt==1.1.0
bs4==0.0.1
bsblan==0.4.0
bt-proximity==0.2.1
bthomehub5-devicelist==0.1.1
btlewrap==0.1.0
btsmarthub-devicelist==0.2.0
buienradar==1.0.5
cached-property==1.5.2
cachetools==5.0.0
caldav==0.7.1
casttube==0.2.1
cattrs==1.10.0
cbor2==5.4.2
cchardet==2.1.7
certifi==2021.10.8
cffi==1.14.6
cfgv==3.3.1
charset-normalizer==2.0.10
circuit-webhook==1.0.1
ciscomobilityexpress==0.3.9
ciso8601==2.2.0
clearpasspy==1.0.2
click==8.0.3
click-log==0.3.2
click-plugins==1.1.1
clx-sdk-xms==1.0.0
[... 959 lines cut ...]
xmltodict==0.12.0
xs1-api-client==3.0.0
yalesmartalarmclient==0.3.4
yalexs==1.1.19
yarl==1.6.3
yeelight==0.7.8
yeelightsunflower==0.0.10
youless-api==0.16
youtube-dl==2021.6.6
zeep==4.1.0
zengge==0.2
zeroconf==0.38.1
zha-quirks==0.0.65
zhong-hong-hvac==1.0.9
ziggo-mediabox-xl==1.1.0
zigpy==0.42.0
zigpy-deconz==0.14.0
zigpy-xbee==0.14.0
zigpy-zigate==0.7.3
zigpy-znp==0.6.4
zipp==3.7.0
zm-py==0.5.2
zope.interface==5.4.0
zwave-js-server-python==0.33.0
```

gieljnssns, 2022-01-22:
> Do you have any other integrations that also use modbus?

I use https://github.com/binsentsu/home-assistant-solaredge-modbus and for now I'm using the modus integration from HA for my Huawei inverter

> How did you add the plugin to your HA?

I have followed this steps

gieljnssns, 2022-01-22:
When I restart HA without the standard modes integration I have this error when trying to install huawei_solar
```
Logger: homeassistant.config_entries
Source: config_entries.py:720
First occurred: 11:24:52 (1 occurrences)
Last logged: 11:24:52

Error occurred loading configuration flow for integration huawei_solar: No module named 'pymodbus.client.asynchronous'
```

gieljnssns, 2022-01-22:
Time ago I also used the custom_component Emilv2 but I had cpu issues https://github.com/Emilv2/huawei_solar/issues/15
That's why I started the integrated modes component
Now I wan't to try yours because the integrated sometimes becomes unavailable and you are using the config flow.

But you and Emil both use `"domain": "huawei_solar"` can this be the problem?

wlcrs, 2022-01-22:
> > Do you have any other integrations that also use modbus?
> 
> I use https://github.com/binsentsu/home-assistant-solaredge-modbus and for now I'm using the modus integration from HA for my Huawei inverter
> 
> > How did you add the plugin to your HA?
> 
> I have followed this steps

Looking at https://github.com/binsentsu/home-assistant-solaredge-modbus/blob/master/custom_components/solaredge_modbus/manifest.json , I see that this integration requires a much lower version of pyModbus. This can cause the issues that you are seeing: HA installs the lower version when it loads that integration, then my integration crashes because of that. That would also explain all the strange errors about missing files, etc.

wlcrs, 2022-01-22:
> Time ago I also used the custom_component Emilv2 but I had cpu issues [Emilv2/huawei_solar#15](https://github.com/Emilv2/huawei_solar/issues/15) That's why I started the integrated modes component Now I wan't to try yours because the integrated sometimes becomes unavailable and you are using the config flow.
> 
> But you and Emil both use `"domain": "huawei_solar"` can this be the problem?

Using the two integrations next to each other is impossible. Apart from the clashing domain: the inverter only allows one connection. Each integration tries to maintain it's own connection, so they would constantly steal it from each other and cause and endless series of disconnects

gieljnssns, 2022-01-22:
`pip freeze` `pymodbus==2.5.3`

> Using the two integrations next to each other is impossible.

I don't want to use them together... but isn't there something left over from that installation?

wlcrs, 2022-01-22:
Can you please disable the solar edge component, reinstall mine and try again? Home assistant will always try to install the dependencies upon startup. This can explain the errors about missing files: they are temporarily removed between the startup of the solaredge and this component.

gieljnssns, 2022-01-27:
Sorry for the late response, but I didn't find the time to test.
I've disabled the solaredge component and removed the core modbus component and removed your component,  restarted HA reinstalled your component, restarted HA again.
[image]

When I try to load your component 

```
Logger: homeassistant.config_entries
Source: config_entries.py:720
First occurred: 15:07:45 (1 occurrences)
Last logged: 15:07:45

Error occurred loading configuration flow for integration huawei_solar: No module named 'pymodbus.client.asynchronous'
```

wlcrs, 2022-01-27:
Hi Giel,

Something is going wrong with the installation of the dependencies on your system. This clearly is not an issue with this library, as many other users have successfully installed it.

If the pymodbus library is correctly installed, then that module should be available.

You can test the following:

```
$ pip uninstall pymodbus
Found existing installation: pymodbus 2.5.3
Uninstalling pymodbus-2.5.3:
  Would remove:
    ....
Proceed (Y/n)? y
  Successfully uninstalled pymodbus-2.5.3
$ pip install --no-cache-dir pymodbus==2.5.3
Collecting pymodbus==2.5.3
  Downloading pymodbus-2.5.3-py2.py3-none-any.whl (154 kB)
     |████████████████████████████████| 154 kB 1.0 MB/s
...
Installing collected packages: pymodbus
Successfully installed pymodbus-2.5.3
$ python3
Python 3.10.2 (main, Jan 17 2022, 19:28:57) [GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pymodbus.client.asynchronous.async_io
>>> dir(pymodbus.client.asynchronous.async_io)
['AsyncModbusClientMixin', 'AsyncioModbusSerialClient', 'AsyncioModbusTcpClient', 'AsyncioModbusUdpClient', 'BaseModbusAsyncClientProtocol', 'ConnectionException', 'DGRAM_TYPE', 'FifoTransactionManager', 'ModbusClientProtocol', 'ModbusUdpClientProtocol', 'ReconnectingAsyncioModbusTcpClient', 'ReconnectingAsyncioModbusTlsClient', 'ReconnectingAsyncioModbusUdpClient', '__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__path__', '__spec__', '__warningregistry__', '_logger', 'asyncio', 'functools', 'hexlify_packets', 'init_tcp_client', 'init_tls_client', 'init_udp_client', 'logging', 'socket', 'ssl']
>>>

```

gieljnssns, 2022-01-28:
Yesterday evening I've updated HA to the new beta while modbus(integrated) and SolarEdge_modbus was disabled. When HA came back to life I tried to install your component, then it worked. 

Thanks for your help.

closed 2022-01-28 by gieljnssns