wlcrs/huawei_solar #19: [Feature Request]: Update interval from 1 to 60 seconds

gugaspinto, 2022-01-29:
### Describe your feature request

The user can't set the update interval. 
During integration installation the user take the possibility to define the time in seconds between inverter data request. 

It will  be perfect if the user can selected from 1 to 60 seconds

There are different hardware HA Server, with very different capabilities and power.

Some of us want to have all sensors information... But other ones only want power yield, and grid power to generate quick information for automations.

If this possibility is possible, any one can adjust this integration for want they need... 

Thanks for your EXCELLENT work... 
Big help for all community

wlcrs, 2022-01-30:
My goal is to get this integration accepted as an official Home Assistant integration. As this integration is already quite large to begin with, I try to keep it as simple as possible (while the approval process is going on).

I've reworked some things to move some of the complexity into the huawei-solar library (cfr.  https://github.com/wlcrs/huawei_solar/commit/6be60b85ff565324003bb4398c4e3e47e2a4877b and https://gitlab.com/Emilv2/huawei-solar/-/commit/eddbefd124cd49c9392d6be5c12ef0b77c5ee5d0 ).  While doing this, I've set the update interval to 30 seconds for everything. 

This reworking will cause some entities to be renamed. So you will need to remove and re-add the integration to get rid of the old 'unavailable' entities. That's why I'm keeping it into the develop-branch for now.

In the meantime, you can change the update intervals yourself by editing the `const.py` file.

gugaspinto, 2022-01-30:
Thanks... Already changed for 2seconds...
It's working...
Greatest integration

closed 2022-01-30 by gugaspinto