wlcrs/huawei_solar #20: [Feature Request]: Add Grid Power Maximum injection

gugaspinto, 2022-01-31:
### Describe your feature request

Writing in the right Modbus Adress, to control the maximum power injected by the inverter in the Grid network.

It will be useful to limit the yield when the energy production is more than the needs. 

For those who not sell the energy... And don't have batteries... This will increase the system longevity. 

Thanks for this Great HA integration

wlcrs, 2022-01-31:
I haven't seen such a register in the documentation. Can you point me to it?

gugaspinto, 2022-01-31:
[Interface.Definitions.V3.0.pdf](https://github.com/wlcrs/huawei_solar/files/7974429/Interface.Definitions.V3.0.pdf)

I think it could be:

Address 40129 (item67, see page 15)

or address 40125 (item70)
percentual difined

or address 40126 (item71)
power (W) difined
**I think this one is better**

wlcrs, 2022-02-01:
Ok, I forgot that I left out those registers as they felt a bit too 'dangerous' / 'useless' to implement. Terminology like 'Q-U characteristic curve mode' made me skip that entire section 😅

Do you know what is the difference between 
[image]
and
[image]

? 

Can you check in the FusionSolar app which setting is used? Provide some screenshots for context.

I think it's better that we clarify this before I just blindly implement this config parameter.

wlcrs, 2022-02-01:
Thanks for your input. However, it doesn't give me a definitive answer on which registers are actually used by FusionSolar. I would need to do some packet capturing again to figure that out.

Is this a setting that you want to regularly change? Because it seems to me that your use case is just to 'set and forget' it?

gugaspinto, 2022-02-02:
For me, it's very interesting, especially in summer.
with this i can create a automation to run:
swimming pool pump
heat the bath water
washing machine
and after all... adjust the yield to the needs
(i have no battery)

I made some inconclusive tests
with HA modbus i made this:
[image]

after that, i changed the export limit from "unlimited" to "1500w" with fusionsolar commissioning

unfortunately, nothing changed... I think there are other Address to make it
I don't see any different ones in huawei modbus interface definitions

I would like to have it...
THANKs

wlcrs, 2022-02-06:
I've done some packet inspection of what the app sends to the inverter, and got the following:
Setting the maximum % to 95%: 
[image]

(this hex value is equivalent to 950)

Setting the maximum kW to 1234:

[image]

(not sure what this value represents... I would guess an I32 value?)

I'm also seeing writes to register 47415 with different values: 0, 7 (I think when using the % setting?), 6 (when using the kW setting? )

All these registers are not properly documented. As this is quite a sensitive setting(*) to change, I'm not willing to 'blindly' implement and test these values on my inverter, let alone risk messing up other users their installations.

(*) The FusionSolar app asks for the password again, and shows an additional confirmation-screen every time you change this.

I'm closing this issue for now, you may reopen it when should these registers appear in future versions of the "Huawei Modbus Register Definitions" file.

wlcrs, 2022-07-13:
**Update:** I found a more recent [Solar Inverter Modbus Interface Definitions.pdf dated 2022-03-29](https://github.com/wlcrs/huawei_solar/files/9102429/Solar.Inverter.Modbus.Interface.Definitions.pdf) . In this document, the registers mentioned here are properly described:

[image]

gugaspinto, 2022-07-14:
Cool .. thanks....
I can try this address with direct Modbus read... Change the parameters with fusion solar and see if address value change...
Is it useful?

wlcrs, 2022-08-22:
I added the following services in beta-release 1.1.3b1:

- `huawei_solar.reset_maximum_feed_grid_power`
- `huawei_solar.set_maximum_feed_grid_power`
- `huawei_solar.set_maximum_feed_grid_power_percent`

Can you test if this feature works as expected in this new release?

cmatte, 2022-09-24:
I can confirm the values shown coincide with the Fusion solar/Solar2000 app. Thanks for integrating this!

closed 2022-09-24 by wlcrs

Vicente-Ch, 2022-09-26:
> Puedo confirmar que los valores mostrados coinciden con la aplicación FusionSolar/Solar2000. ¡Gracias por integrar esto!

I have the sun2000-6ktl-l1 and connected to the home assistant through an Elfin-EW11 by rtu.
 What inverter do you have and firmware version

Vicente-Ch, 2022-09-26:
Hello, has anyone tried it with a single-phase sun2000 6KTL L1 inverter.  Does the software version of the inverter matter?  (I have the V200R001C00SPC119)
Thanks

jodewee, 2024-06-20:
@gugaspinto , @wlcrs is this the whole inverter's output that will be limited, or just the feed to the grid?

Vicente-Ch, 2024-06-21:
> @gugaspinto,@wlcrs¿Se limitará toda la producción del inversor o solo la alimentación a la red?

Yo no lo he probado, pues cuando hice la instalacion no active la opcion "avanzado". Para activar esto hay q borrar la integracion y volver a instalarla?

lenniepennie, 2025-05-20:
When I set "huawei_solar.set_maximum_feed_grid_power" service, it limits the whole inverters output. I would expect that it just limits the power to the grid. What do I do wrong?