wlcrs/huawei_solar #22: [Bug]: Doesn't appear in the Add Integration list

ShaneGooseman, 2022-02-03:
### System Health details

Home Assistant 2022.2.0

### Describe the issue

i install the component with HACS, reboot HA but then it does not appear in the list of available integrations

### Reproduction steps

1.Install this integration with HACS,
2.Restart HA
3.Go to Configuration -> Integrations and click the + Add Integration button
4.No puedo seleccionar Huawei Solar de la lista, no aparece

### Relevant debug logs

```text
as
```

wlcrs, 2022-02-03:
If you don't provide better debug logs, I will close this issue as there is nothing for me to work with.

ShaneGooseman, 2022-02-03:
2022-02-03 14:28:53 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2022-02-03 14:28:53 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration huawei_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant

ShaneGooseman, 2022-02-03:
> If you don't provide better debug logs, I will close this issue as there is nothing for me to work with.

I update the issue

wlcrs, 2022-02-03:
Can you show me the contents of your `/config/custom_components/huawei_solar/manifest.json` file?

Does it appear if you do a hard refresh of your HA-page? (CTRL+F5 for most browsers on Windows/Linux).

ShaneGooseman, 2022-02-03:
Hello again.
I went to eat and when I came back Huawei Solar appeared in the list of integrations.
I don't know if it was due to some internal HA process or something in the browser.

closed 2022-02-03 by wlcrs