wlcrs/huawei_solar #23: [Bug]: Possible integration broken after upgrading to HA 2022.02

inigoml, 2022-02-06:
### System Health details

## System Health

version | core-2022.2.2
-- | --
installation_type | Home Assistant Supervised
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 4.19.0-10-amd64
arch | x86_64
timezone | Europe/Madrid

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4813
Installed Version | 1.22.0
Stage | running
Available Repositories | 980
Downloaded Repositories | 5

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Debian GNU/Linux 11 (bullseye)
-- | --
update_channel | stable
supervisor_version | supervisor-2022.01.1
docker_version | 20.10.12
disk_total | 105.4 GB
disk_used | 36.9 GB
healthy | true
supported | true
supervisor_api | ok
version_api | ok
installed_addons | Home Assistant Google Drive Backup (0.105.2), Studio Code Server (4.1.0), SQLite Web (3.2.1)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 0
views | 7
mode | storage

</details>

<details><summary>Xiaomi Miot Auto</summary>

component_version | 0.5.2
-- | --
can_reach_server | ok
can_reach_spec | ok
logged_accounts | 1
total_devices | 57

</details>

<details><summary>Xiaomi MIoT</summary>

logged_in | true
-- | --
can_reach_micloud_server | ok
accounts_count | 1
account_devices_count | 57
added_devices | 7

</details>

### Describe the issue

Integration does not start.

### Reproduction steps

1. Start Home Assistant (hassio_supervised docker edition)
2. Integration does not start

### Relevant debug logs

```text
This error originated from a custom integration.

Logger: homeassistant.config_entries
Source: custom_components/huawei_solar/__init__.py:40
Integration: Huawei Solar
First occurred: 5:52:11 PM (1 occurrences)
Last logged: 5:52:11 PM

Error setting up entry SUN2000-4KTL-L1 for huawei_solar
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 327, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/huawei_solar/__init__.py", line 40, in async_setup_entry
    primary_bridge = await HuaweiSolarBridge.create(
TypeError: create() got an unexpected keyword argument 'username'
```

inigoml, 2022-02-06:
Right now I'm working with login-and-write brach. Will test with master branch

inigoml, 2022-02-06:
Switching back to main branch solved the issue (but lost writing capabilities).
It appears that the problem is based at HuaweiSolarBridge class. Perhaps this HA version uses a different huawei-solar class.

bash-5.1# pip3 list | grep huawei
huawei-lte-api                   1.4.18
huawei-solar                     2.0.0rc1

inigoml, 2022-02-06:
[image]

wlcrs, 2022-02-06:
Hi, I continued working on the login-and-write branch of both the integration and the library, which caused things to get out of sync for you. I've now removed that branch of the integration, as I felt that it was ready to be moved into `develop`.

Can you please test again with v1.0.0a1 which I just released? You will likely need to remove and add the integration again to get the configuration done properly. I've only considered the scenario of people who were on my main branch.

inigoml, 2022-02-07:
Testing. Working again.
I see you have changed the name of "active_power_2" sensor to "grid_active_power." It makes sense!

JacoboSanchez, 2022-02-07:
> Testing. Working again. I see you have changed the name of "active_power_2" sensor to "grid_active_power." It makes sense!

[Reference on names discussion](https://github.com/wlcrs/huawei_solar/issues/21) (it goes a bit offtopic after closing it :)

closed 2022-02-07 by wlcrs