wlcrs/huawei_solar #24: [Bug]: Issue activating the Integration

taert79, 2022-02-07:
### System Health details

## System Health

version | core-2022.2.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.88
arch | x86_64
timezone | Europe/Brussels

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4515
Installed Version | 1.22.0
Stage | running
Available Repositories | 978
Downloaded Repositories | 11

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 4 March 2022, 01:00
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | eu-west-2-0.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.1
-- | --
update_channel | stable
supervisor_version | supervisor-2022.01.1
docker_version | 20.10.9
disk_total | 118.5 GB
disk_used | 9.4 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Studio Code Server (4.1.0), Grafana (7.4.1), InfluxDB (4.3.0), Terminal & SSH (9.3.0), Mopidy (6e49c79)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 4
views | 1
mode | storage

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Describe the issue

I receive an error when trying to install the Integration (see debug log below)

### Reproduction steps

1. Install via HACS
2. Restart HA
3. Try to install the Integration

### Relevant debug logs

```text
Logger: aiohttp.server
Source: requirements.py:194
First occurred: 11:16:57 (1 occurrences)
Last logged: 11:16:57

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_protocol.py", line 435, in _handle_request
    resp = await request_handler(request)
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 60, in security_filter_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 98, in forwarded_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 79, in ban_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 219, in auth_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 137, in handle
    result = await result
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 132, in post
    return await super().post(request)
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 62, in wrapper
    result = await method(view, request, *args, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 72, in post
    result = await self._flow_mgr.async_init(
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 203, in async_init
    flow, result = await task
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 221, in _async_init
    flow = await self.async_create_flow(handler, context=context, data=data)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 730, in async_create_flow
    await async_process_deps_reqs(self.hass, self._hass_config, integration)
  File "/usr/src/homeassistant/homeassistant/setup.py", line 351, in async_process_deps_reqs
    await requirements.async_get_integration_with_requirements(
  File "/usr/src/homeassistant/homeassistant/requirements.py", line 83, in async_get_integration_with_requirements
    await _async_process_integration(hass, integration, done)
  File "/usr/src/homeassistant/homeassistant/requirements.py", line 99, in _async_process_integration
    await async_process_requirements(
  File "/usr/src/homeassistant/homeassistant/requirements.py", line 162, in async_process_requirements
    await _async_process_requirements(
  File "/usr/src/homeassistant/homeassistant/requirements.py", line 194, in _async_process_requirements
    raise RequirementsNotFound(name, [req])
homeassistant.requirements.RequirementsNotFound: Requirements for huawei_solar not found: ['huawei-solar==2.1.0rc4'].
```

wlcrs, 2022-02-07:
I had tagged/released an incorrect version '0.5.0'. I have pulled that one, and created a correct release instead.

If you are installing this integration for the first time, I would recommend trying the v1.0.0 alpha 1. This will cause you less issues (with sensors that are renamed) down the line when the integration moves to core. The 'alpha' part is only for the "parameter configuation" part, reading sensors should be very stable!

closed 2022-02-07 by wlcrs

taert79, 2022-02-07:
Thank you Thijs! Working perfectly now.