wlcrs/huawei_solar #26: [Bug]: No labels for fields to install the integration

johanvh83, 2022-02-10:
### System Health details

System Health

version: core-2022.2.5
installation_type: Home Assistant OS
dev: false
hassio: true
docker: true
user: root
virtualenv: false
python_version: 3.9.7
os_name: Linux
os_version: 5.10.91
arch: x86_64
timezone: Europe/Brussels

GitHub API: ok
Github API Calls Remaining: 4517
Installed Version: 1.22.0
Stage: running
Available Repositories: 1043
Downloaded Repositories: 24

logged_in: false
can_reach_cert_server: ok
can_reach_cloud_auth: ok
can_reach_cloud: ok

host_os: Home Assistant OS 7.2
update_channel: stable
supervisor_version: supervisor-2022.01.1
docker_version: 20.10.9
disk_total: 30.8 GB
disk_used: 17.9 GB
healthy: true
supported: true
board: ova
supervisor_api: ok
version_api: ok
installed_addons: Duck DNS (1.14.0), Samba share (9.5.1), Studio Code Server (4.1.0), Mosquitto broker (6.0.1), Node-RED (10.4.0), Zigbee2mqtt (1.23.0-1), Home Assistant Google Drive Backup (0.105.2), ESPHome (2022.1.3), Grafana (7.4.1), InfluxDB (4.3.0), Frigate NVR (2.3)

dashboards: 1
resources: 15
views: 14
mode: storage

api_endpoint_reachable: ok

### Solar Setup Details

Single phase
Power meter used
Inverter SUN2000-4.6KTL-L1  with firmware V200R001C00SPC115
10 kWh Luna battery with firmware V100R002C00SPC106

### Describe the issue

Unable to install the integration due to missing labels for the input fields.

### Reproduction steps

1. Install your integration by HACS
2. Restarted HA
3. Installed your integration in HA (Configuration - Devices & Services - Integrations - + Add integration - and got a first popup with 3 input fields but without labels:
[image]
I guess the first field is the IP address. Second field is I guess the por (I left it on the default value) and the 3th field represents the slave option. No idea if the tickbox is used for optimisers or battery but I enabled the option.

After this I get another popup with 2 fields:

[image]

The first box shows 'intaller' and cannot be changed.
The second box looks like a password box. I entered my FusionSolar login but got an error. 
Please instruct what to fill in the second box.

Thanks in advance.

### Relevant debug logs

```text
no logs
```

wlcrs, 2022-02-10:
Hi Johan,

Strange that these strings are missing for you, as I cannot replicate the issue on a fresh HA-install:

[image]

[image]

The second screen asks after the credentials used to login on the inverter itself with the "Device commissioning" tool. The default password is `00000a`.  I don't think that any installer bothers with changing that password. 

However, as I do consider it 'dangerous' to allow changing the battery parameters for an uneducated user, I prefer to explicitly ask for that password.

Were you using version 1.0.0a2?

johanvh83, 2022-02-11:
Hi I tried once more with the default HA theme and used your beta version 1.0.0a3. I could setup the integration with all labels displayed. Unluckaly couldn't proceed as it seems my electrician changed the login credentials.
I asked him the credentials. Fingers crossed.
Alternatively I installed V0.5.1. All works but I got some strange values; Grid active power shows 2,147,483,647 W.

wlcrs, 2022-02-11:
If you leave the 'advanced: enable parameter configuration' checkbox unchecked, you will not need to insert the installer credentials. Can you let me know if the strange values are still present in that case? If so: can you please list all sensors which have the wrong value?

wlcrs, 2022-02-11:
Also: you can reset the installer password by following the steps listed here: https://support.huawei.com/enterprise/en/doc/EDOC1100136173/8aa1f88a/resetting-password

johanvh83, 2022-02-11:
1/ To clarify: you want me to install the beta version 1.0.0a3 with the checkbox unchecked? 
2/ I guess I need the installer credentials to be able to manage the charging of the battery. 
3/ I wait for my electricians feedback. If he doesn't want to share the credentials, I perform a reset ;-)

wlcrs, 2022-02-11:
1/ Yes

johanvh83, 2022-02-11:
Before I change to 1.0.0a3 I'd like to inform you that the 'Daily Yield' sensor cannot be used for the Solar production within the Energy Integration. I did it that way but the result is incorrect.

johanvh83, 2022-02-11:
Hi with 1.0.0a3 the Grid Active Power sensor shows a correct value (positive). It reprents now the injected power to the grid.

closed 2022-02-11 by wlcrs