wlcrs/huawei_solar #27: [Bug]:

barto64, 2022-02-11:
### System Health details

## System Health

version | core-2022.2.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.98
arch | aarch64
timezone | Europe/Madrid

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 5000
Installed Version | 1.22.0
Stage | running
Available Repositories | 986
Downloaded Repositories | 34

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.4
-- | --
update_channel | stable
supervisor_version | supervisor-2022.01.1
docker_version | 20.10.9
disk_total | 113.9 GB
disk_used | 13.2 GB
healthy | true
supported | true
board | odroid-n2
supervisor_api | ok
version_api | ok
installed_addons | Samba share (9.5.1), Check Home Assistant configuration (3.9.0), File editor (5.3.3), NGINX Home Assistant SSL proxy (3.1.1), Portainer (1.5.0), Duck DNS (1.14.0), Terminal & SSH (9.3.0), Mosquitto broker (6.0.1), BlueRiiot2MQTT (0.14.0), motionEye (0.12.0), Traccar (0.12.0), ESPHome (1.17.1), MariaDB (2.4.0), Z-Wave JS (0.1.53)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 18
views | 12
mode | yaml

</details>

<details><summary>Spotify</summary>

[... 72 lines cut ...]
2022-02-11 17:30:59 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.1.91', 6607)
2022-02-11 17:30:59 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Waiting 800 ms before next connection attempt.
2022-02-11 17:31:00 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 43006
2022-02-11 17:31:00 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2022-02-11 17:31:00 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting.
2022-02-11 17:31:00 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.1.91', 6607)
2022-02-11 17:31:00 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Waiting 1600 ms before next connection attempt.
2022-02-11 17:31:02 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting.
2022-02-11 17:31:02 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.1.91', 6607)
2022-02-11 17:31:02 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Waiting 3200 ms before next connection attempt.
2022-02-11 17:31:05 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting.
2022-02-11 17:31:05 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.1.91', 6607)
2022-02-11 17:31:05 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Waiting 6400 ms before next connection attempt.
2022-02-11 17:31:11 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting.
2022-02-11 17:31:11 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.1.91', 6607)
2022-02-11 17:31:11 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Waiting 12800 ms before next connection attempt.
2022-02-11 17:31:11 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting.
2022-02-11 17:31:11 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.1.91', 502)
2022-02-11 17:31:11 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Waiting 51200 ms before next connection attempt.
2022-02-11 17:31:19 ERROR (MainThread) [homeassistant.components.shelly] Timeout fetching Shelly Puerta Valla data
2022-02-11 17:31:24 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting.
2022-02-11 17:31:24 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.1.91', 6607)
2022-02-11 17:31:24 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Waiting 25600 ms before next connection attempt.
```

wlcrs, 2022-02-11:
Your dongle is not allowing connections.

Please check in the 'Device Comissioning' app in Settings --> Communication Settings if everything is properly configured. The modbus communication must be set to "Unrestricted". You can post screenshots if you are doubting your settings.

Please let me know what (combination of) settings fixed it for you. Then I can add it to the documentation: screenshots are easy and helpful!

barto64, 2022-02-12:
These are my settings, the dongel has enabled de modbus tcp without restrictions. I cannot understand why it does not allows connections.

[image]
[image]
[image]

barto64, 2022-02-12:
As a side note, my installation is made of two inverters which are connected 
[image]

wlcrs, 2022-02-12:
You've checked the settings on both inverters?

I don't have much more advice to give. I have an installation without
SDongle (I connect to the SUNxxxxx wifi), and have only one inverter. So no
practical experience in getting setups like these working properly.

You could ask on
https://community.home-assistant.io/t/integration-solar-inverter-huawei-2000l/132350/last
for some more advice.

barto64, 2022-02-12:
Yes, I checked. Great thanks anyway. Will go to the community.

closed 2022-02-12 by wlcrs

gieljnssns, 2022-02-12:
@barto64 
Can you try to add your dongle to your WiFi. 
I had also problems with modbus over Ethernet.

barto64, 2022-02-12:
I did it and I managed to connect apparently although the integration still saying "Failed to connect". New log below:

2022-02-12 13:50:28 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting to 192.168.1.92:502.
2022-02-12 13:50:28 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting.
2022-02-12 13:50:28 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client connected to modbus server
2022-02-12 13:50:28 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol made connection.
2022-02-12 13:50:28 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Connected to 192.168.1.92:502.
2022-02-12 13:50:28 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client disconnected from modbus server: None
2022-02-12 13:50:28 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol lost connection.
2022-02-12 13:50:28 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Waiting 100 ms before next connection attempt.
2022-02-12 13:50:29 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting.
2022-02-12 13:50:29 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client connected to modbus server
2022-02-12 13:50:29 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol made connection.
2022-02-12 13:50:29 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Connected to 192.168.1.92:502.
2022-02-12 13:50:29 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client disconnected from modbus server: None
2022-02-12 13:50:29 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol lost connection.
2022-02-12 13:50:29 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Waiting 100 ms before next connection attempt.
2022-02-12 13:50:29 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting.
2022-02-12 13:50:29 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client connected to modbus server
2022-02-12 13:50:29 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol made connection.
2022-02-12 13:50:29 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Connected to 192.168.1.92:502.
2022-02-12 13:50:29 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client disconnected from modbus server: None
2022-02-12 13:50:29 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol lost connection.
2022-02-12 13:50:29 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Waiting 100 ms before next connection attempt.
2022-02-12 13:50:29 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting.
2022-02-12 13:50:29 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client connected to modbus server
2022-02-12 13:50:29 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol made connection.
2022-02-12 13:50:29 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Connected to 192.168.1.92:502.
2022-02-12 13:50:29 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client disconnected from modbus server: None
2022-02-12 13:50:29 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol lost connection.
2022-02-12 13:50:29 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Waiting 100 ms before next connection attempt.
2022-02-12 13:50:29 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting.
2022-02-12 13:50:29 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client connected to modbus server
2022-02-12 13:50:29 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol made connection.
2022-02-12 13:50:29 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Connected to 192.168.1.92:502.
2022-02-12 13:50:29 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client disconnected from modbus server: None
2022-02-12 13:50:29 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol lost connection.
2022-02-12 13:50:29 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Waiting 100 ms before next connection attempt.
2022-02-12 13:50:29 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting.
2022-02-12 13:50:29 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client connected to modbus server
2022-02-12 13:50:29 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol made connection.
2022-02-12 13:50:29 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Connected to 192.168.1.92:502.
2022-02-12 13:50:29 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client disconnected from modbus server: None
2022-02-12 13:50:29 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol lost connection.
2022-02-12 13:50:29 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Waiting 100 ms before next connection attempt.
2022-02-12 13:50:29 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 43006
2022-02-12 13:50:29 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2022-02-12 13:50:29 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting.
2022-02-12 13:50:29 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client connected to modbus server
2022-02-12 13:50:29 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol made connection.
2022-02-12 13:50:29 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Connected to 192.168.1.92:502.
[... cut ...]
2022-02-12 13:50:32 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Connected to 192.168.1.92:502.
2022-02-12 13:50:32 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client disconnected from modbus server: None
2022-02-12 13:50:32 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol lost connection.
2022-02-12 13:50:32 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Waiting 100 ms before next connection attempt.
2022-02-12 13:50:32 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting.

wlcrs, 2022-02-12:
You are almost there!

That behavior occurs when you have multiple clients competing for access.
Make sure to use 1.0.0a3 as it fixes some scenarios where old attempts
become zombies that keep on retrying in the background. It's best to do a
full power cycle of your HA machine to kill those zombies.

wlcrs, 2022-02-13:
https://community.home-assistant.io/t/integration-solar-inverter-huawei-2000l/132350/1235#how-to-test-3