wlcrs/huawei_solar #28: [Bug]: Permission denied after lost of connection with the inverter

JacoboSanchez, 2022-02-16:
### System Health details

## System Health

version | core-2022.2.7
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.98
arch | x86_64
timezone | Europe/Madrid

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4862
Installed Version | 1.22.0
Stage | running
Available Repositories | 1048
Downloaded Repositories | 23

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.4
-- | --
update_channel | stable
supervisor_version | supervisor-2022.01.1
docker_version | 20.10.9
disk_total | 234.0 GB
disk_used | 12.0 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (9.3.0), Mosquitto broker (6.0.1), AppDaemon 4 (0.8.1), Studio Code Server (4.1.0), ESPHome (2022.1.4), SQLite Web (3.2.1), WireGuard (0.6.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 3
-- | --
resources | 20
views | 6
mode | storage

</details>

### Describe the issue

When there is a connection issue at the inverter (last one was due to a network issue at my house) then authenticated connection that allows writing a register get lost and any further modification produces a Permission denied error

### Reproduction steps

1. Connect the integration to the inverter
2. Wait until there is some connection failure to the inverter (for example battery SOC goes to 0 at some point or forcing network issues)
3. Try to write a register
...

Permission denied error which is fixed restarting HA

### Relevant debug logs

```text
2022-02-16 10:01:52 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection] [139737392269776] Received {'type': 'call_service', 'domain': 'number', 'service': 'set_value', 'service_data': {'value': '49', 'entity_id': 'number.grid_charge_cutoff_soc'}, 'id': 65}
2022-02-16 10:01:52 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=number, service=set_value, service_data=value=49, entity_id=number.grid_charge_cutoff_soc>
2022-02-16 10:01:52 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection] [139737487853360] Sending {"id": 48, "type": "event", "event": {"event_type": "call_service", "data": {"domain": "number", "service": "set_value", "service_data": {"value": "49", "entity_id": "number.grid_charge_cutoff_soc"}}, "origin": "LOCAL", "time_fired": "2022-02-16T09:01:52.733201+00:00", "context": {"id": "4c554543b3f15ddb84b61428af1a8a06", "parent_id": null, "user_id": "43afe331828341e092825df35208db49"}}}
2022-02-16 10:01:52 DEBUG (MainThread) [pymodbus.payload] [490]
2022-02-16 10:01:52 DEBUG (MainThread) [pymodbus.payload] [490]
2022-02-16 10:01:52 DEBUG (MainThread) [huawei_solar.huawei_solar] Writing to 47088: [490]
2022-02-16 10:01:52 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0x24 0x0 0x0 0x0 0x9 0x0 0x10 0xb7 0xf0 0x0 0x1 0x2 0x1 0xea
2022-02-16 10:01:52 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 36
2022-02-16 10:01:52 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x0 0x24 0x0 0x0 0x0 0x3 0x0 0x90 0x80
2022-02-16 10:01:52 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x24 0x0 0x0 0x0 0x3 0x0 0x90 0x80
2022-02-16 10:01:52 DEBUG (MainThread) [pymodbus.factory] Factory Response[144]
2022-02-16 10:01:52 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 36
2022-02-16 10:01:52 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [139737392269776] Permission denied
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 190, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1630, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1667, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 204, in handle_service
    await self.hass.helpers.service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 668, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 930, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 705, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 75, in async_set_value
    await entity.async_set_value(value)
  File "/config/custom_components/huawei_solar/number.py", line 197, in async_set_value
    if await self.bridge.set(self.entity_description.key, int(value)):
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/bridge.py", line 240, in set
    return await self.client.set(name, value, slave=self.slave_id)
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 304, in set
    response = await self.write_registers(
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 335, in write_registers
    raise PermissionDenied("Permission denied")
huawei_solar.exceptions.PermissionDenied: Permission denied
2022-02-16 10:01:52 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection] [139737392269776] Sending {"id": 65, "type": "result", "success": false, "error": {"code": "unknown_error", "message": "Permission denied"}}
```

JacoboSanchez, 2022-03-25:
I switched to modbus-rtu connection with the inverter so I am not using huawei modbus-tcp connection any more. My guess is that I was affected by loss of connection frequently due to the router having many wifi devices but in my opinion this makes the inverter's configuration not reliable to use in automations.

wlcrs, 2022-05-05:
Duplicated by #57

closed 2022-05-05 by wlcrs