wlcrs/huawei_solar #29: [Bug]: Huawei sensors are going unavailable sometimes

gieljnssns, 2022-02-16:
### System Health details

## System Health

version | core-2022.2.7
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.98
arch | x86_64
timezone | Europe/Brussels

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 5000
Installed Version | 1.22.0
Stage | running
Available Repositories | 1060
Downloaded Repositories | 36

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.4
-- | --
update_channel | beta
supervisor_version | supervisor-2022.01.1
docker_version | 20.10.9
disk_total | 457.7 GB
disk_used | 112.5 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Samba share (9.5.1), AdGuard Home (4.4.4), AppDaemon 4 (0.8.1), Caddy (1.6), Check Home Assistant configuration (3.9.0), Duck DNS (1.14.0), ESPHome (2022.1.4), Mosquitto broker (6.0.1), Studio Code Server (4.1.0), WireGuard (0.6.0), MariaDB (2.4.0), Samba Backup (5.0.0), InfluxDB (4.3.0), Grafana (7.4.1), Frigate NVR (2.3), SSH & Web Terminal (10.0.2), Glances (0.14.2), phpMyAdmin (0.7.0), Mealie (0.5.6), Arpspoof (1.0.0), Caddy 2 (1.1.0), Firefly iii (5.6.14-2)

</details>

<details><summary>Lovelace</summary>

dashboards | 3
-- | --
resources | 17
views | 18
mode | storage

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Describe the issue

My HA instance is connected to the internal wifi of the inverter

All Huawei sensors are going unavailable for some time.
[image]

[image]

### Reproduction steps

1. Connect to internal wifi
2. Install integration
3. Just wait
...

### Relevant debug logs

```text
Deze fout is ontstaan door een aangepaste integratie.

Logger: custom_components.huawei_solar
Source: helpers/update_coordinator.py:193
Integration: Huawei Solar
First occurred: 11:40:50 (1 occurrences)
Last logged: 11:40:50

Timeout fetching HV20C0092844_data_update_coordinator data
```

wlcrs, 2022-02-16:
Thank you for providing the logs. I see frequent connection drops, do you have good connectivity to your inverter? Is the device connecting to the inverter WiFi close enough? Maybe you can run a ping to the inverter to check if you see the connection drops there too?

I'll increase the timeout values in an upcoming release, but I'm not sure that this will do much for you as the connections seem to drop entirely, and the Modbus-TCP implementation of Huawei does not handle this well.
The integration periodically tries to reconnect (with a maximum of 5 seconds between attempts).

What inverter do you have?

gieljnssns, 2022-02-16:
I have an sun2000-4ktl-m1 inverter with firmware version V100R001C00SPC141 
It is located about 1,5m of the intel nuc I'm using for HA. There is no wall in between them.

gieljnssns, 2022-02-17:
Indeed with ping I also have bad results
[image]

wlcrs, 2022-02-17:
Yikes. Not much I can do in such a situation. I suggest you look into improving stability of your connection. I'll also be documenting the usage of the physical RS485_1 Modbus-interface on the future. You might consider using that instead.

gieljnssns, 2022-02-17:
@wlcrs 
I have downgraded my inverter to V100R001C00SPC135 again. Then the ping times where normal again. But the modbus port changed to 502 again. 
There is no way to change te modbus port in your integration. So I updated to V100R001C00SPC140. 
modbus port is 6607 again. And ping responses are normal. 
[image]

wlcrs, 2022-02-18:
You can change the port by removing and reinstalling the integration

gugaspinto, 2022-02-22:
Hello
It happens to me... Unavailable sensors
But only during 3 seconds... Can say, Around every 1h30m

I'm connected over internal WiFi, with good coverage, with good ping, good connection.
I'm using the stable version of the huawei_solar, I think it's 0.05 (after installing, I don't see the version... Sorry) 

I only have 5 sensors activated on huawei_solar integration, but I update values every second. 

I have changed from dongle LAN to internal WiFi, because I sow anywhere its faster... 

My Log says SLAVEBUSY, as you can see above. 

For my automations is important to have 1sec values.
In Portugal we pay the electricity with a 15min balance, and my HA calculate that balance to make the automation triggers

There are some way to solve this? 

Talking about performance... There are any difference between this integration and the HA Modbus? 

Thanks for your GREAT WORK... Congratulations

[image]

wlcrs, 2022-02-22:
You are trying to read faster from the inverter than it can handle, hence the SlaveBusy errors. Nothing I can change about this.

Changing to the native HA Modbus integration will not solve this: you will just get the same SlaveBusy errors, without the advantage of the multi-register-reads that this integration performs. 

In the beta versions I automatically retry on SlaveBusy errors.

gugaspinto, 2022-02-22:
Reducing the number of activated entities in the integration will reduce the errors? 
For example, only the power yield 1 and 2

Do you know if the rs485 is faster? 
I don't have the necessary hardware... Rs2lan
If it is a solution I could think about that... 

Thanks for your patience
🙂

wlcrs, 2022-02-22:
Sensors are bundled: one inverter bundle, one power meter bundle, one
battery bundle, one pc string bundle. Enabling a sensor of one bundle will
query all datafields of that bundle, regardless of how many are actually
enabled.

I have not tried using the RS485_1 interface directly myself (yet). Any
feedback would be appreciated on the performance of that interface vs using
the WLAN or dongle.

tfriberg, 2022-08-16:
I built a template to manage when the inverter is _unresponsive_, or _turned off_ at night.

```
template:
  sensors:
    template_inverter_active_power_availability: 
      friendly_name: Inverter active power
      device_class: energy
      unit_of_measurement: 'W'
      value_template: '{{ 0 if states ("sensor.inverter_active_power") == "unavailable" else states ("sensor.inverter_active_power") }}'
```
[image]

This gives me live updates (and live daily yield numbers when the inverter is unreachable) on the front page (the gauges using sensor_inverter_active_power reports 0 when unavailable and does not replace the gauges with an error message). 
The calculations I do to present what the house consumes at the moment (solar power PLUS from grid MINUS to grid) is another template sensor;

```
    template_active_power_consumption:
      friendly_name: Power consumption
      device_class: energy
      unit_of_measurement: 'W'
      value_template: '{{ states ("sensor.inverter_active_power")|int(0) + states ("sensor.lgf_active_power_import") | int(0) - states ("sensor.lgf_active_power_export")|int(0) }}'
```

Hope this helps @gieljnssns.

It would, however, be very nice if the inverter reported 0 "out-of-the-box" when not reachable.

wlcrs, 2022-08-16:
I understand that it would be nice to have those entities report as 0, but that is not easy to implement: returning 0 for some registers would cause all kinds of problems and unintended consequences (monotonically increasing registers like total yield for example). 

I think that your approach with templates is the best way forward to be honest.

gugaspinto, 2022-08-17:
The better solution is never have "unavailable" moments 

What's your data update frequency?
What kind of connection do you have with the inverter?

I collect data every second.
I have some "unavailable" moments... I can say... 30 times a day...

I think it  became worth when the integration updated to have rs485

I connect through dongle Lan directly to my router.
I already tried inverter WiFi to my AP --> router. Same result

For me, it's impossible to have rs485 usb to my rpi.
Can rs485 to Lan do a better result?
Thanks

stefanhapper, 2022-11-23:
@gugaspinto How do you manage to query the inverter every second? Even if update the inverter values every 10 seconds the inverter stops sending data to Huawei FusionSolar. Or you don't care that no data is sent to FusionSolar?

I have a SUN200 4.6 KTL-L1 with SPC124 and connect via AP wifi

Pixelalb, 2025-03-18:
https://take.ms/ZqKbK

Since yesterday, all sensors from the Huawei Solar integration in Home Assistant go unavailable simultaneously for a short period before coming back online. This behavior did not occur before yesterday when I set up a Modbus connection between my Nibe S320 heat pump and the Huawei inverter.

I have attached a graph showing the issue’s start point. The sensors were stable before the Modbus connection was configured.

Would appreciate any insights from the community on potential causes or fixes!