wlcrs/huawei_solar #30: [Bug]: Error while trying to add integration of version 1.0.0a3

alexoman2030, 2022-02-18:
### System Health details

## System Health

version | core-2022.2.5
-- | --
installation_type | Home Assistant Core
dev | false
hassio | false
docker | false
user | root
virtualenv | true
python_version | 3.9.8
os_name | Linux
os_version | 5.13.0-30-generic
arch | x86_64
timezone | UTC

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4844
Installed Version | 1.22.0
Stage | running
Available Repositories | 1043
Downloaded Repositories | 2

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Lovelace</summary>

dashboards | 2
-- | --
resources | 0
views | 3
mode | storage

</details>

### Huawei Solar Setup

1 x Sun2000 KTL 4.6 L1 (Connected by wire to Main router. IP inverter : 192.168.0.209)
1 x Luna2000 10kWh
1 x 3-phase power meter

Laptop with Ubuntu running HA connected to a Secondary Router by Wifi
Laptop IP  (IP 192.168.1.118)  
DNS 192.168.0.1

Both routers are connected by wire together

### Describe the issue

Hello, 
I'm pretty new in the HA world (playing since 2 weeks with it, and waiting for the Raspberry Pi 4 B 4Gb to arrive very soon)

I was able to install the Version 0.5.1 of Huawei Solar.
I wanted to install the 1.0.0a3  to be on the latest version with the optimisation that I could read in the forum, but when I do this, I get the Error message "Config flow could not be loaded: [object Object]"
-> I tried to uninstall in HACS
-> Clear cache and cookie from browser
-> Close Broswer
-> Restart HA
-> Update HA
-> HACS seems to be already on the latest version

### Reproduction steps

1.Install Huawei Solar from HACS
2.Restart HA Server management via Configuration> Settings
3. Clear Browser cache & Cookie
4.Try to add the integration Huawei Solar from Configuration>Devices & Services
5. Loading a while then Error
6. Retry
7. Call for support :)

### Relevant debug logs

```text
Logger: homeassistant.config_entries
Source: config_entries.py:735
First occurred: 10:43:36 PM (2 occurrences)
Last logged: 10:54:02 PM

Error occurred loading configuration flow for integration huawei_solar: cannot import name 'HuaweiSolarBridge' from 'huawei_solar' (/root/snap/home-assistant-snap/483/.local/lib/python3.9/site-packages/huawei_solar/__init__.py)

2022-02-18 22:43:17 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration huawei_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2022-02-18 22:43:17 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2022-02-18 22:43:19 WARNING (MainThread) [homeassistant.components.updater]
NOTE! Using a replacement (custom) 'updater' component for the snap package.
Do NOT report bugs of any kind related to 'updater' to the Home Assistant core project.
Report any issues at https://github.com/home-assistant-snap/home-assistant-snap/issues
2022-02-18 22:43:19 WARNING (MainThread) [homeassistant.components.updater]
NOTE! Using a replacement (custom) 'updater' component for the snap package.
Do NOT report bugs of any kind related to 'updater' to the Home Assistant core project.
Report any issues at https://github.com/home-assistant-snap/home-assistant-snap/issues
2022-02-18 22:43:19 WARNING (MainThread) [homeassistant.components.meteo_france] 1 hour rain forecast not available. Halle - Flanders - BE is not in covered zone
2022-02-18 22:43:19 WARNING (MainThread) [homeassistant.components.meteo_france] Weather alert not available: The city Halle - Flanders - BE is not in metropolitan France or Andorre
2022-02-18 22:43:23 WARNING (MainThread) [homeassistant.components.homeassistant.triggers.numeric_state] Error initializing 'Turn on boiler' trigger: In 'numeric_state' condition: unknown entity sensor.input_power
2022-02-18 22:43:23 WARNING (MainThread) [homeassistant.components.homeassistant.triggers.numeric_state] Error initializing 'Turn OFF boiler ' trigger: In 'numeric_state' condition: unknown entity sensor.input_power
2022-02-18 22:43:36 ERROR (MainThread) [homeassistant.config_entries] Error occurred loading configuration flow for integration huawei_solar: cannot import name 'HuaweiSolarBridge' from 'huawei_solar' (/root/snap/home-assistant-snap/483/.local/lib/python3.9/site-packages/huawei_solar/__init__.py)
2022-02-18 22:54:02 ERROR (MainThread) [homeassistant.config_entries] Error occurred loading configuration flow for integration huawei_solar: cannot import name 'HuaweiSolarBridge' from 'huawei_solar' (/root/snap/home-assistant-snap/483/.local/lib/python3.9/site-packages/huawei_solar/__init__.py)
2022-02-18 23:01:10 ERROR (MainThread) [homeassistant.components.upnp] Timeout fetching WRT320N data
```

wlcrs, 2022-02-19:
The supporting library `huawei-solar` was not correctly installed/updated by HA (as defined in the `manifest.json`).

Can you find any errors in the logs during the installation of the integration?

alexoman2030, 2022-02-19:
Do you mean to check the logs after having downloaded the Huawei_Solar integration from HACS?
-> If not, how do I access those logs?

Warning after Huawei Solar integration from HACS
Logger: homeassistant.loader
Source: loader.py:344
First occurred: 9:02:00 AM (3 occurrences)
Last logged: 9:04:45 AM

We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
We found a custom integration huawei_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant 

logs after HA restart. 

2022-02-19 09:08:09 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration huawei_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2022-02-19 09:08:09 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2022-02-19 09:08:10 WARNING (MainThread) [homeassistant.components.updater]
NOTE! Using a replacement (custom) 'updater' component for the snap package.
Do NOT report bugs of any kind related to 'updater' to the Home Assistant core project.
Report any issues at https://github.com/home-assistant-snap/home-assistant-snap/issues
2022-02-19 09:08:11 WARNING (MainThread) [homeassistant.components.updater]
NOTE! Using a replacement (custom) 'updater' component for the snap package.
Do NOT report bugs of any kind related to 'updater' to the Home Assistant core project.
Report any issues at https://github.com/home-assistant-snap/home-assistant-snap/issues
2022-02-19 09:08:11 WARNING (MainThread) [homeassistant.components.meteo_france] 1 hour rain forecast not available. Halle - Flanders - BE is not in covered zone
2022-02-19 09:08:11 WARNING (MainThread) [homeassistant.components.meteo_france] Weather alert not available: The city Halle - Flanders - BE is not in metropolitan France or Andorre
2022-02-19 09:08:21 WARNING (MainThread) [homeassistant.config_entries] Config entry 'WRT320N' for upnp integration not ready yet; Retrying in background
2022-02-19 09:08:21 WARNING (MainThread) [homeassistant.components.homeassistant.triggers.numeric_state] Error initializing 'Turn on boiler' trigger: In 'numeric_state' condition: unknown entity sensor.input_power
2022-02-19 09:08:21 WARNING (MainThread) [homeassistant.components.homeassistant.triggers.numeric_state] Error initializing 'Turn OFF boiler ' trigger: In 'numeric_state' condition: unknown entity sensor.input_power

logs After trying to add integration in the config>Devices&Services

File "/snap/home-assistant-snap/483/lib/python3.9/site-packages/aiohttp/web_app.py", line 504, in _handle
resp = await handler(request)
File "/snap/home-assistant-snap/483/lib/python3.9/site-packages/aiohttp/web_middlewares.py", line 117, in impl
return await handler(request)
File "/snap/home-assistant-snap/483/lib/python3.9/site-packages/homeassistant/components/http/security_filter.py", line 60, in security_filter_middleware
return await handler(request)
File "/snap/home-assistant-snap/483/lib/python3.9/site-packages/homeassistant/components/http/forwarded.py", line 98, in forwarded_middleware
return await handler(request)
File "/snap/home-assistant-snap/483/lib/python3.9/site-packages/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
return await handler(request)
File "/snap/home-assistant-snap/483/lib/python3.9/site-packages/homeassistant/components/http/ban.py", line 79, in ban_middleware
return await handler(request)
File "/snap/home-assistant-snap/483/lib/python3.9/site-packages/homeassistant/components/http/auth.py", line 219, in auth_middleware
return await handler(request)
File "/snap/home-assistant-snap/483/lib/python3.9/site-packages/homeassistant/components/http/view.py", line 137, in handle
result = await result
File "/snap/home-assistant-snap/483/lib/python3.9/site-packages/homeassistant/components/config/config_entries.py", line 222, in post
return await super().post(request, flow_id)
File "/snap/home-assistant-snap/483/lib/python3.9/site-packages/homeassistant/components/http/data_validator.py", line 62, in wrapper
result = await method(view, request, *args, **kwargs)
File "/snap/home-assistant-snap/483/lib/python3.9/site-packages/homeassistant/helpers/data_entry_flow.py", line 111, in post
result = await self._flow_mgr.async_configure(flow_id, data)
File "/snap/home-assistant-snap/483/lib/python3.9/site-packages/homeassistant/data_entry_flow.py", line 252, in async_configure
result = await self._async_handle_step(flow, cur_step["step_id"], user_input)
File "/snap/home-assistant-snap/483/lib/python3.9/site-packages/homeassistant/data_entry_flow.py", line 325, in _async_handle_step
result: FlowResult = await getattr(flow, method)(user_input)
File "/snap/home-assistant-snap/483/lib/python3.9/site-packages/homeassistant/components/upnp/config_flow.py", line 295, in async_step_init
coordinator = self.hass.data[DOMAIN][self.config_entry.entry_id]
KeyError: 'a571d3bcf306c0cab9a6be3293be1303'
[... cut ...]
    result = await self._async_handle_step(flow, cur_step["step_id"], user_input)
  File "/snap/home-assistant-snap/483/lib/python3.9/site-packages/homeassistant/data_entry_flow.py", line 325, in _async_handle_step
    result: FlowResult = await getattr(flow, method)(user_input)
  File "/snap/home-assistant-snap/483/lib/python3.9/site-packages/homeassistant/components/upnp/config_flow.py", line 295, in async_step_init
    coordinator = self.hass.data[DOMAIN][self.config_entry.entry_id]
KeyError: 'a571d3bcf306c0cab9a6be3293be1303'

wlcrs, 2022-02-22:
The logs in your latest post don't show that you're installing the huawei_solar integration.

Please fully remove the integration, restart HA, install it again, and post the logs you get after doing that.

Home Assistant should install the correct library version of the "huawei-solar" python library, but didn't do that in your case. This is a bug in HA, and not in the integration I'm afraid.

closed 2022-02-28 by wlcrs