wlcrs/huawei_solar #31: [Feature Request]: Battery status entity

joaovitoriasilva, 2022-02-22:
### Describe your feature request

Hello,

Is it possible to have a entity to give the battery status, like running, stand by, etc?

Thank you.

wlcrs, 2022-02-22:
I'll include a sensor for the following register in the next release:

```
  rn.STORAGE_RUNNING_STATUS: U16Register(rv.StorageStatus, 1, 37762, 1),
```

```
class StorageStatus(IntEnum):
    """Status of the attached energy storage."""

    OFFLINE = 0
    STANDBY = 1
    RUNNING = 2
    FAULT = 3
    SLEEP_MODE = 4
```

joaovitoriasilva, 2022-02-22:
Awesome! Thank you. 🙂

VederJay, 2022-03-06:
Seems to be working now, thanks!

wlcrs, 2022-03-06:
I still need to fix the values so that they are better "human readable".
I'll keep this issue open for now until that is fixed

wlcrs, 2022-05-05:
Has been improved in 1.0.1

closed 2022-05-05 by wlcrs