wlcrs/huawei_solar #33: [Bug]:

2bbe, 2022-02-25:
### System Health details

## System Health

version | core-2022.2.9
-- | --
installation_type | Home Assistant Container
dev | false
hassio | false
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.4.72-microsoft-standard-WSL2
arch | x86_64
timezone | Europe/Stockholm

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4810
Installed Version | 1.23.0
Stage | running
Available Repositories | 997
Downloaded Repositories | 32

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Lovelace</summary>

dashboards | 3
-- | --
resources | 15
views | 25
mode | storage

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Huawei Solar Setup

Inverter	SUN2000-6KTL-M1		V100R001C00SPC140
Dongle	SDongleA 			V100R001C00SPC126

### Describe the issue

The efficiency value goes from 0 to 655,35% before production starts and right after production stops. See image.
This happens mainly during standby:detecting irradiation and standby:init
It looks like an overflow for unsigned short due to division by zero.

[image]

### Reproduction steps

See Above

### Relevant debug logs

```text
To be added if needed
```

wlcrs, 2022-02-25:
This indeed looks like a bug, but one of the inverter firmware :-/ I double checked the register definitions PDF to make sure that I've implemented the register correctly. 

I'm not very inclined to start fixing all sorts of inverter-bugs by adding extra logic in the library, as this increases maintenance, and adds a layer of "magic" (and thus confusion/support requests) which I would rather avoid.

I suggest you add a "template sensor" which filters out the wrong values.

2bbe, 2022-02-25:
I understand.

closed 2022-02-25 by 2bbe