wlcrs/huawei_solar #35: [Bug]: Got error while reading from register 37760 with length 28: Exception Response(131, 3, IllegalAddress)

eicosan, 2022-03-02:
### System Health details

## System Health

version | core-2022.2.9
-- | --
installation_type | Home Assistant Supervised
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.0-11-amd64
arch | x86_64
timezone | Europe/Berlin

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4906
Installed Version | 1.23.0
Stage | running
Available Repositories | 1071
Downloaded Repositories | 4

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Debian GNU/Linux 11 (bullseye)
-- | --
update_channel | stable
supervisor_version | supervisor-2022.01.1
docker_version | 20.10.12
disk_total | 32.7 GB
disk_used | 5.9 GB
healthy | true
supported | true
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.3.3), Node-RED (11.0.4), Mosquitto broker (6.0.1), Ring-MQTT with Video Streaming (4.9.1)

</details>

<details><summary>Lovelace</summary>

dashboards | 2
-- | --
resources | 0
views | 2
mode | storage

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Huawei Solar Setup

Model
SUN2000-6KTL-M0 
Software version
V100R001C00SPC133 

Model
SDongleA-01 
Software version
V100R001C00SPC122 

Power Meter - 3 phase

no battery

Raspberry Pi is configured to be a network bridge between inverters AP (wlan0 192.168.8.0/24) and local home network (eth0 192.168.2.0/24). On Home assistant server the route is configured. Ping to inverter works, telnet to port 502 works, a self written python script (same server) can read registers 37113 and 32080 with unit id = 1. Script disabled for testing this integration to not conflict.

### Describe the issue

After adding the integration (1.0.0a3) it confirms that the inverter was successfully added but the integration shows following error: 
Could not update HV1970006XXX values: Got error while reading from register 37760 with length 28: Exception Response(131, 3, IllegalAddress)

### Reproduction steps

1. Add repository / custom component via HACS
2. Restart HAAS 
3. Add integration Huawei Solar4. 
4. Provide values IP: 192.168.8.1, Port: 502, Slave ID: 0 (also tried 1 and 2)

### Relevant debug logs

```text
2022-03-02 10:05:22 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol made connection.
2022-03-02 10:05:22 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Connected to 192.168.8.1:502.
2022-03-02 10:05:23 INFO (MainThread) [custom_components.huawei_solar.config_flow] Successfully connected to inverter SUN2000-6KTL-M0 with SN HV1970006XXX
2022-03-02 10:05:23 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol lost connection.
2022-03-02 10:05:23 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol made connection.
2022-03-02 10:05:23 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Connected to 192.168.8.1:502.
2022-03-02 10:05:25 WARNING (MainThread) [homeassistant.config_entries] Config entry 'SUN2000-6KTL-M0' for huawei_solar integration not ready yet: Could not update HV1970006XXX v
alues: Got error while reading from register 37760 with length 28: Exception Response(131, 3, IllegalAddress); Retrying in background
2022-03-02 10:05:25 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol lost connection.
2022-03-02 10:05:30 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol made connection.
2022-03-02 10:05:30 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Connected to 192.168.8.1:502.
2022-03-02 10:05:31 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol lost connection.
2022-03-02 10:05:41 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol made connection.
2022-03-02 10:05:41 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Connected to 192.168.8.1:502.
2022-03-02 10:05:43 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol lost connection.
2022-03-02 10:06:03 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol made connection.
2022-03-02 10:06:03 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Connected to 192.168.8.1:502.
2022-03-02 10:06:05 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol lost connection.
```

eicosan, 2022-03-02:
I think the mentioned register 37760 is battery soc, but I do not have a battery...

wlcrs, 2022-03-02:
Hi Johannes, thanks for your well written bugreport!

I try to auto-detect if a battery is present, but apparently your inverter behaves differently from what I expected that would happen. I'll fix the detection mechanism.

eicosan, 2022-03-02:
Thanks wlcrs! 

Additional debug log if that helps...

`2022-03-02 13:08:06 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 12
2022-03-02 13:08:07 DEBUG (MainThread) [pymodbus.payload] [0, 2326, 0, 2338, 0, 2336, 0, 242, 0, 229, 0, 222, 0, 1277, 0, 617, 64748, 5000, 15, 12656, 9, 58455, 0, 0, 1, 0, 4040, 0, 4047, 0, 4037, 0, 542, 0, 466, 0, 269]
2022-03-02 13:08:07 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\t\x16']
2022-03-02 13:08:07 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\t"']
2022-03-02 13:08:07 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\t ']
2022-03-02 13:08:07 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x00\xf2']
2022-03-02 13:08:07 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x00\xe5']
2022-03-02 13:08:07 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x00\xde']
2022-03-02 13:08:07 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x04\xfd']
2022-03-02 13:08:07 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x02i']
2022-03-02 13:08:07 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x0f', b'1p']
2022-03-02 13:08:07 DEBUG (MainThread) [pymodbus.payload] [b'\x00\t', b'\xe4W']
2022-03-02 13:08:07 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x00\x00']
2022-03-02 13:08:07 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x0f\xc8']
2022-03-02 13:08:07 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x0f\xcf']
2022-03-02 13:08:07 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x0f\xc5']
2022-03-02 13:08:07 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x02\x1e']
2022-03-02 13:08:07 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x01\xd2']
2022-03-02 13:08:07 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x01\r']
2022-03-02 13:08:07 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-03-02 13:08:07 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0xd 0x0 0x0 0x0 0x6 0x0 0x3 0x93 0x80 0x0 0x1c
2022-03-02 13:08:07 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 13
2022-03-02 13:08:07 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x0 0xd 0x0 0x0 0x0 0x3 0x0 0x83 0x2
2022-03-02 13:08:07 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0xd 0x0 0x0 0x0 0x3 0x0 0x83 0x2
2022-03-02 13:08:07 DEBUG (MainThread) [pymodbus.factory] Factory Response[131]
2022-03-02 13:08:07 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 13
2022-03-02 13:08:07 WARNING (MainThread) [homeassistant.config_entries] Config entry 'SUN2000-6KTL-M0' for huawei_solar integration not ready yet: Could not update HV1970006950 values: Got error while reading from register 37760 with length 28: Exception Response(131, 3, IllegalAddress); Retrying in background
2022-03-02 13:08:07 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client disconnected from modbus server: None`

wlcrs, 2022-03-02:
Can you check if it is fixed in 1.0.0a4 for you?

eicosan, 2022-03-02:
I was able to successfully enable the integration, now! 
Thanks for the fast fix 👍

closed 2022-03-02 by eicosan

DanieleCar, 2023-12-27:
Hello , i have the same problem of this bug , with really same error :  error while reading from register 37760 with length 28: Exception Response(131, 3, IllegalAddress) 
my configuration : 

Inverter Type: Sun2000L 5ktl
Inverter Firmware version: V100R001C00SPC335
SDongle present: no 
Power meter present: single phase 
Battery: LG RESU 9.8kWh 
Battery Firmware version: na 
[home-assistant_huawei_solar_2023-12-27T10-16-43.714Z.log](https://github.com/wlcrs/huawei_solar/files/13777807/home-assistant_huawei_solar_2023-12-27T10-16-43.714Z.log)

Thanks