wlcrs/huawei_solar #36: Connecting to this integration

JensPauwels, 2022-03-04:
Hi

I read that this repository is the improved version of the huawei solar from emilv2. 
I tried to replace the emilv2 huawei solar with this one, however i couldn't manage to get it connected. Is the connection procedure different to the one from emilv2?

I have a sun2000 3tkl m1 which is connected with a s dongle.

[INVERTER] = spc140
[DONGLE] = spc124

Thanks in advance!

JacoboSanchez, 2022-03-04:
Posting logs would help to see what fails in your scenario. At the end both integrations use modbus-tcp so they need the same information (ip, port, slave_id...) there is no reason for one to work but the other fail. Did you removed the previous configuration? inverter does only support one client at a time so you can not use  both integrations or one integration and also connect with the fusionsolar app.
Also what version did you install? If you used version 0.5.1 I would suggest to reinstall enabling beta versions and to try with version 1.0.0a4

JensPauwels, 2022-03-04:
I'm kind of a beginner with home assistant. So i'll look where to find the logs. I did indeed remove the huawei solar from emil before installing this one. I installed version 0.5.1, where can i find the beta version?

JensPauwels, 2022-03-04:
The beta versions don't show up in my home assistant. Trying to use main branch now

JacoboSanchez, 2022-03-04:
Try going here in HACS:
[image]

And then tick the beta versions:

[image]

Of course langage would not be the same in your environment

JensPauwels, 2022-03-04:
2022-03-04 20:59:08 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client connected to modbus server
2022-03-04 20:59:08 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol made connection.
2022-03-04 20:59:08 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client disconnected from modbus server: None
2022-03-04 20:59:08 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol lost connection.
2022-03-04 20:59:08 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Connected to 192.168.5.25:502.
2022-03-04 20:59:08 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Waiting 100 ms before next connection attempt.
2022-03-04 20:59:08 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting.
2022-03-04 20:59:08 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client connected to modbus server
2022-03-04 20:59:08 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol made connection.
2022-03-04 20:59:08 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Connected to 192.168.5.25:502.
2022-03-04 20:59:08 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client disconnected from modbus server: None
2022-03-04 20:59:08 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol lost connection.
2022-03-04 20:59:08 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Waiting 100 ms before next connection attempt.

These are the logs i see at the moment

JacoboSanchez, 2022-03-04:
Can't really help you there, did you try to change slave_id from 0 to 1?

JensPauwels, 2022-03-04:
Thank you! It succeeded

closed 2022-03-04 by JensPauwels