wlcrs/huawei_solar #37: [Bug]: Connection drops

eicosan, 2022-03-07:
### System Health details

## System Health

version | core-2022.3.2
-- | --
installation_type | Home Assistant Supervised
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.10.0-11-amd64
arch | x86_64
timezone | Europe/Berlin

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4968
Installed Version | 1.23.0
Stage | running
Available Repositories | 1079
Downloaded Repositories | 5

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Debian GNU/Linux 11 (bullseye)
-- | --
update_channel | stable
supervisor_version | supervisor-2022.01.1
docker_version | 20.10.12
disk_total | 32.7 GB
disk_used | 10.6 GB
healthy | true
supported | true
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.3.3), Node-RED (11.0.4), Mosquitto broker (6.0.1), Ring-MQTT with Video Streaming (4.9.1), Duck DNS (1.14.0), Studio Code Server (4.1.0), ESPHome (2022.2.6)

</details>

<details><summary>Lovelace</summary>

dashboards | 3
-- | --
resources | 1
views | 9
mode | storage

</details>

<details><summary>Spotify</summary>
[... 65 lines cut ...]
2022-03-07 19:30:30 INFO (MainThread) [homeassistant.components.automation.huawei_solar_inverter_data_polling] Huawei Solar inverter data polling: Running automation actions
2022-03-07 19:30:30 INFO (MainThread) [homeassistant.components.automation.huawei_solar_inverter_data_polling] Huawei Solar inverter data polling: Executing step call service
...
...
2022-03-07 19:45:37 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (asyncio.exceptions.TimeoutError)
2022-03-07 19:45:42 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (asyncio.exceptions.TimeoutError)
2022-03-07 19:45:45 INFO (MainThread) [homeassistant.components.automation.huawei_solar_inverter_data_polling] Huawei Solar inverter data polling: Running automation actions
2022-03-07 19:45:45 INFO (MainThread) [homeassistant.components.automation.huawei_solar_inverter_data_polling] Huawei Solar inverter data polling: Executing step call service
2022-03-07 19:45:47 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (asyncio.exceptions.TimeoutError)
2022-03-07 19:45:50 WARNING (MainThread) [homeassistant.helpers.entity] Update of media_player.spotify_johannes_richter is taking over 10 seconds
2022-03-07 19:45:52 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (asyncio.exceptions.TimeoutError)
2022-03-07 19:45:56 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol lost connection.
2022-03-07 19:45:58 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol made connection.
2022-03-07 19:45:58 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Connected to 192.168.8.1:502.
2022-03-07 19:46:00 INFO (MainThread) [homeassistant.components.automation.huawei_solar_inverter_data_polling] Huawei Solar inverter data polling: Running automation actions
2022-03-07 19:46:00 INFO (MainThread) [homeassistant.components.automation.huawei_solar_inverter_data_polling] Huawei Solar inverter data polling: Executing step call service
2022-03-07 19:46:04 INFO (MainThread) [custom_components.huawei_solar] Fetching HV197000XXXX_data_update_coordinator data recovered
2022-03-07 19:46:11 ERROR (SyncWorker_10) [spotipy.client] Max Retries reached
2022-03-07 19:46:15 INFO (MainThread) [homeassistant.components.automation.huawei_solar_inverter_data_polling] Huawei Solar inverter data polling: Running automation actions
2022-03-07 19:46:15 INFO (MainThread) [homeassistant.components.automation.huawei_solar_inverter_data_polling] Huawei Solar inverter data polling: Executing step call service
2022-03-07 19:46:30 INFO (MainThread) [homeassistant.components.automation.huawei_solar_inverter_data_polling] Huawei Solar inverter data polling: Running automation actions
2022-03-07 19:46:30 INFO (MainThread) [homeassistant.components.automation.huawei_solar_inverter_data_polling] Huawei Solar inverter data polling: Executing step call service
2022-03-07 19:46:45 INFO (MainThread) [homeassistant.components.automation.huawei_solar_inverter_data_polling] Huawei Solar inverter data polling: Running automation actions
2022-03-07 19:46:45 INFO (MainThread) [homeassistant.components.automation.huawei_solar_inverter_data_polling] Huawei Solar inverter data polling: Executing step call service
```

wlcrs, 2022-03-07:
This is an issue of the SDongle firmware and not of the integration. I don't get a timely response, so the integration retries the request.

I'm planning on adding support for the direct RS485_1 port in the long term, which hopefully does not have this problem, but will require a wired connection between the HA-device and the inverter.

eicosan, 2022-03-07:
Anyway, is it possible to close and reconnect in that case? 
In the logs you can see that after the connection was lost and re-established it works again.

wlcrs, 2022-03-07:
I said SDongle because I read reports from other owners that said that it behaves even more poorly than the built-in WiFi. But in any case: it's more of a bug on the inverter side.

You're welcome to contribute a "service call" which forces a reconnect. However, as this would not be allowed in an official HA integration (even setting a custom refresh interval configuration option is not allowed), I'm not going to develop it myself.

eicosan, 2022-03-08:
> You're welcome to contribute a "service call" which forces a reconnect. However, as this would not be allowed in an official HA integration (even setting a custom refresh interval configuration option is not allowed), I'm not going to develop it myself.

Unfortunately, I have no idea how this can be done. If you can point me in the right direction...
Can I do it via an automation checking the status and in case data is not available do something (reconnect modbus)?

eicosan, 2022-03-09:
@wlcrs, thanks for your tip. I was able to solve the problem by myself. I've implemented a rest call as a shell command that reloads the integration. This shell command is executed via an automation when a one of the Huawei sensors is unavailable for some time.

closed 2022-03-09 by eicosan