wlcrs/huawei_solar #38: [Bug]:

dml33, 2022-03-17:
### System Health details

## System Health

version | core-2022.3.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.10.92-v8
arch | aarch64
timezone | Europe/Madrid

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4919
Installed Version | 1.23.0
Stage | running
Available Repositories | 1081
Downloaded Repositories | 6

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.5
-- | --
update_channel | stable
supervisor_version | supervisor-2022.03.5
docker_version | 20.10.9
disk_total | 58.7 GB
disk_used | 8.2 GB
healthy | true
supported | true
board | rpi3-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.3.3), SSH & Web Terminal (10.1.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 2
-- | --
resources | 1
views | 1
mode | storage

</details>

### Huawei Solar Setup

Huawei Solar Integration: v0.5.1
Host: 192.168.200.1
Port: 6607
Slave: 0
SUN2000-3KTL-L1. Firmware 117
Power sensor single phase 
No dongle
No battery
Connected to 192.168

### Describe the issue

After updating from 2022.03.1, the Huawei Solar Integration has stopped working.

[image]

### Reproduction steps

1. It was working perfectly for some weeks with Home Assistant 2022.3.1
2. Update Home Assistant from 2022.3.1 to 2022.3.5
3. It does not work at all. It appears as "not loaded"
...

### Relevant debug logs

```text
This is the error in the log:

This error originated from a custom integration.

Logger: aiohttp.server
Source: custom_components/huawei_solar/__init__.py:10
Integration: Huawei Solar
First occurred: 21:23:55 (3 occurrences)
Last logged: 21:25:43

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_protocol.py", line 435, in _handle_request
    resp = await request_handler(request)
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 60, in security_filter_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 100, in forwarded_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 79, in ban_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 219, in auth_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 137, in handle
    result = await result
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 72, in delete
    result = await hass.config_entries.async_remove(entry_id)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 858, in async_remove
    await entry.async_remove(self.hass)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 510, in async_remove
    component = integration.get_component()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 553, in get_component
    cache[self.domain] = importlib.import_module(self.pkg_path)
  File "/usr/local/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/custom_components/huawei_solar/__init__.py", line 10, in <module>
    from huawei_solar import (
ModuleNotFoundError: No module named 'huawei_solar'

```
```

wlcrs, 2022-03-17:
> timezone Europe/Madrid Home Assistant Community Store
> Downloaded Repositories 6 Home Assistant Cloud
> can_reach_cloud ok Home Assistant Supervisor
> installed_addons File editor (5.3.3), SSH & Web Terminal (10.1.0) Lovelace
> mode storage Huawei Solar Setup

Can you try to reinstall the integration? The error tells me that the
library on which the integration depends is not installed anymore. This is
a bug of HA, and not of the integration.

If the issue persists, please include the logs just after reinstalling.
This should contain more info about why the library was not correctly
installed.

dml33, 2022-03-17:
Thanks for the reply.

The thing is that I cannot delete it even. 

I go to Configuration -> Devices&Services -> click on the three dots on the Huawei Solar, click on Delete, click on "OK", I am sure, but it does not do anything. In the log, the following error appears:

This error originated from a custom integration.

```
Logger: aiohttp.server
Source: custom_components/huawei_solar/__init__.py:10
Integration: Huawei Solar
First occurred: 22:15:19 (1 occurrences)
Last logged: 22:15:19

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_protocol.py", line 435, in _handle_request
    resp = await request_handler(request)
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 60, in security_filter_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 100, in forwarded_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 79, in ban_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 219, in auth_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 137, in handle
    result = await result
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 72, in delete
    result = await hass.config_entries.async_remove(entry_id)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 858, in async_remove
    await entry.async_remove(self.hass)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 510, in async_remove
    component = integration.get_component()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 553, in get_component
    cache[self.domain] = importlib.import_module(self.pkg_path)
  File "/usr/local/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/custom_components/huawei_solar/__init__.py", line 10, in <module>
    from huawei_solar import (
ModuleNotFoundError: No module named 'huawei_solar'
```

dml33, 2022-03-17:
If I try to delete it from HACS, the following error appears:

```
Logger: aiohttp.server
Source: loader.py:553
First occurred: 22:19:44 (1 occurrences)
Last logged: 22:19:44

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_protocol.py", line 435, in _handle_request
    resp = await request_handler(request)
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.9/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 60, in security_filter_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 100, in forwarded_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 79, in ban_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 219, in auth_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 137, in handle
    result = await result
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 72, in delete
    result = await hass.config_entries.async_remove(entry_id)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 858, in async_remove
    await entry.async_remove(self.hass)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 510, in async_remove
    component = integration.get_component()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 553, in get_component
    cache[self.domain] = importlib.import_module(self.pkg_path)
  File "/usr/local/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 984, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'custom_components.huawei_solar'
```

dml33, 2022-03-17:
I reinstalled it from HACS. These are the two errors after restarting HA:

```
Logger: homeassistant.util.package
Source: util/package.py:99
First occurred: 22:23:03 (3 occurrences)
Last logged: 22:23:52

Unable to install package huawei-solar==2.0.0rc1: WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate (_ssl.c:1129)'))': /packages/ad/93/77d96cd21f83510d7e383355c7cf23c1676806a350b7770f345987c70082/huawei_solar-2.0.0rc1-py3-none-any.whl WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate (_ssl.c:1129)'))': /packages/ad/93/77d96cd21f83510d7e383355c7cf23c1676806a350b7770f345987c70082/huawei_solar-2.0.0rc1-py3-none-any.whl WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate (_ssl.c:1129)'))': /packages/ad/93/77d96cd21f83510d7e383355c7cf23c1676806a350b7770f345987c70082/huawei_solar-2.0.0rc1-py3-none-any.whl WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate (_ssl.c:1129)'))': /packages/ad/93/77d96cd21f83510d7e383355c7cf23c1676806a350b7770f345987c70082/huawei_solar-2.0.0rc1-py3-none-any.whl WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate (_ssl.c:1129)'))': /packages/ad/93/77d96cd21f83510d7e383355c7cf23c1676806a350b7770f345987c70082/huawei_solar-2.0.0rc1-py3-none-any.whl ERROR: Could not install packages due to an OSError: HTTPSConnectionPool(host='files.pythonhosted.org', port=443): Max retries exceeded with url: /packages/ad/93/77d96cd21f83510d7e383355c7cf23c1676806a350b7770f345987c70082/huawei_solar-2.0.0rc1-py3-none-any.whl (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate (_ssl.c:1129)'))) WARNING: You are using pip version 22.0.3; however, version 22.0.4 is available. You should consider upgrading via the '/usr/local/bin/python3 -m pip install --upgrade pip' command.
```

And also:

```
Logger: homeassistant.setup
Source: setup.py:162
First occurred: 22:23:52 (1 occurrences)
Last logged: 22:23:52

Setup failed for custom integration huawei_solar: Requirements for huawei_solar not found: ['huawei-solar==2.0.0rc1'].
```

wlcrs, 2022-03-17:
Those errors show that connecting to the pypi registry that hosts the
library fails (SSL error). This causes the installation to fail.

This should hopefully resolve itself. Otherwise it indicates an issue with
the SSL certificates trusted by your HA server 😱

closed 2022-04-29 by wlcrs