wlcrs/huawei_solar #39: [Bug]: Incorrect values after loss of AC power

thtro, 2022-03-22:
### System Health details

## System Health

version | core-2022.3.6
-- | --
installation_type | Home Assistant Supervised
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.10.0-10-amd64
arch | x86_64
timezone | Europe/Bucharest

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4865
Installed Version | 1.23.0
Stage | running
Available Repositories | 1008
Downloaded Repositories | 10

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Debian GNU/Linux 11 (bullseye)
-- | --
update_channel | stable
supervisor_version | supervisor-2022.03.5
docker_version | 20.10.12
disk_total | 218.1 GB
disk_used | 90.0 GB
healthy | true
supported | failed to load: Unsupported
supervisor_api | ok
version_api | ok
installed_addons | Duck DNS (1.14.0), SSH & Web Terminal (10.1.1), Studio Code Server (4.1.0), Mosquitto broker (6.0.1), ESPHome (2022.3.0), NGINX Home Assistant SSL proxy (3.1.1), Samba share (9.5.1), Zigbee2mqtt (1.24.0-1), Transmission (dev), RTSPtoWebRTC (1.0.1), rtl_433 MQTT Auto Discovery (0.3.1), Log Viewer (0.13.0), Node-RED (11.1.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 2
-- | --
resources | 5
views | 5
mode | storage

</details>

<details><summary>Xiaomi Miot Auto</summary>

component_version | 0.5.4
-- | --
can_reach_server | ok
can_reach_spec | ok
logged_accounts | 1
total_devices | 5

</details>

### Huawei Solar Setup

Sun2000 8KTL-M1
Meter - 3 phase
Wifi SDongle
All with latest firmware
no battery

### Describe the issue

After the AC power is lost (due to grid issues i.e.) It seems that a very high number of kwh is reported(as produced and consumed) - which completely screws up the stats
I noticed this issue with the other HASS plugin for Huawei inverters as well, I'm not sure what produces it(I suppose it's a bad reading from the registry via Modbus, maybe an exception should cover it)
 On the Solar Fusion web app all it's normal.
[image]
[image]

### Reproduction steps

1. Use as normal
2. Disconnect the grid completely (including the power meter)
3. Wait for the stats in HASS again.
...

### Relevant debug logs

```text
debugging was not on.
```

wlcrs, 2022-04-29:
Those numbers are very close to the maximum number that can be reported by the inverter. I assume that it returns the maximum number in a register is it has no valid value for it.

I've added a check for this in the underlying library: https://gitlab.com/wlcrs/huawei-solar/-/commit/c4abd451d1fe3a227ff77816b36551f6b4ef303f . This should hopefully fix this issue.

wlcrs, 2022-05-05:
Fixed in 1.0.1 . 

Tip: Since HA 2022.4 you can easily remove such erroneous values from the time series.

closed 2022-05-05 by wlcrs

thtro, 2022-05-05:
Thanks for all the efforts put in maintaining this. All the best!