wlcrs/huawei_solar #41: [Feature Request]: Optimizer Feature Data

brillo83, 2022-04-02:
### Describe your feature request

It would be very interesting to be able to create sensors for individual optimizers with info about Output Power, Total Yield and alarms, too
[image]
. According to Huawei solar Modbus Interface the register No. 59 (Optmizer Feature Data) should contain this data.

wlcrs, 2022-04-03:
Reading the optimizer data on my to-do list. I need to implement a custom modbus protocol to read these values. Note that these are updated at a much lower rate than the other registers.

BertisAU, 2022-05-03:
Very much looking forward to this update! Thank you.

madimadi, 2022-05-29:
As per modbus interface definition document the huawei defined function 0x41 describes the method how to get such data.
After extensive search on its implementation I doubt anyone did that so far.  Do you think the huawei doc is detailed enough to write code on their huwaei defined file upload method?

wlcrs, 2022-05-29:
I've just finished my implementation this weekend. I even reverse-engineered the FusionSolar app to figure out how to request a specific range of optimizer-data (instead of returning all the values for every 5 minutes that the inverter has been live...): https://gitlab.com/Emilv2/huawei-solar/-/commit/0b4c70a5e28984dd23c25c33e17cb7e339780416

I'm now working on updating the integration to expose these values as HA Entities. Hopefully I can put a beta version online to test later today

madimadi, 2022-05-29:
You are awesome. Will give a try of your library soon.
I see v102 version is supported. Do you know it is failing with other version?

wlcrs, 2022-05-29:
I just released a beta version of the integration containing the optimizer feature data: 1.1.0b2 (b1 contained a bug that I discovered while writing the release notes).

Several new entities with identical names will be generated. To differentiate between them, you need to check the device with which each entity is associated. 

Please read the release notes for more information. Please create new issues for any unexpected behavior, as the issue template will help me understand your setup.

closed 2022-05-29 by wlcrs

brillo83, 2022-06-08:
I've just loaded

> I just released a beta version of the integration containing the optimizer feature data: 1.1.0b2 (b1 contained a bug that I discovered while writing the release notes).
>
> Several new entities with [...]

Thanks for this release, i've just updated to 1.1.0b5, the integration added devices for optimizers (12 in my case) but all sensors are not available :(...I checked the advanced option (and typed the installer credentials).
May be it is due to the optmizers are not online now? Hope so

[image]