wlcrs/huawei_solar #45: After update to v1.0.0 - a new interver was added

fcqpl, 2022-04-16:
Hi.
I updated from v0.5.1 to v1.0.0.
This addon added new inverter with same name and "_2" at end entity name.

How to upgrade without changing entity names and all other integrations with it?

## System Health

version | core-2022.3.8
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.10.108
arch | x86_64
timezone | Europe/Warsaw

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4881
Installed Version | 1.24.0
Stage | running
Available Repositories | 1100
Downloaded Repositories | 5

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.6
-- | --
update_channel | stable
supervisor_version | supervisor-2022.04.0
docker_version | 20.10.9
disk_total | 31.3 GB
disk_used | 16.4 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.3.3), Terminal & SSH (9.3.0), Mosquitto broker (6.0.1), Zigbee2mqtt (1.25.0-1), Node-RED (11.1.1)

</details>

<details><summary>Lovelace</summary>

dashboards | 5
-- | --
resources | 1
views | 8
mode | storage

</details>

edvincent, 2022-04-18:
+1, same problem... Added all the sensors `_2` to my Energy dashboard to be able to have the old and the new metrics, but now there are two entries in the tables and different colors etc... which is a bit painful ;(

fcqpl, 2022-04-18:
I restored backup before upgrade and waiting for some update.
Maybe better @edvincent also restore backup before upgrade?

wlcrs, 2022-05-05:
I'm sorry, but with only two reports it is not worthwhile for me to put effort into this issue. Please delete all the old entities manually before upgrading to 1.0.x .

I'd rather put effort into other issues affecting more people, and which cannot be fixed by some manual intervention by the user.

closed 2022-05-05 by wlcrs

fabo137, 2022-05-10:
Have two inverters as well at the moment. How i can delete the old one? At the moment i can only disable it but not remove. 

I have no problems with doing it manuallly, how can I do it without removing the whole integration?

wlcrs, 2022-05-10:
Try the Devices tab instead of the integrations one

fcqpl, 2022-05-10:
Hi,
- Make HA Backup
- Remove integration from Integrations tab
- Upgrade Huawei Solar to 1.X
- Reboot HA
- Add Huawei Solar integration again

You'll have one inverter and old entities.

fabo137, 2022-05-10:
> Hi,
> 
> * Make HA Backup
> * Remove integration from Integrations tab
> * Upgrade Huawei Solar to 1.X
> * Reboot HA
> * Add Huawei Solar integration again
> 
> You'll have one inverter and old entities.

Solve my Problem Thanks.