wlcrs/huawei_solar #47: [Feature Request]: add Time of use settings for battery charge/discharge

jrowland1993, 2022-04-18:
### Describe your feature request

thanks for all your hard work on this integration, it’s brilliant!

I’m not sure if this is possible, but adding the settings for charge discharge of the battery based on time of use (available on the sun2000 app as installer only) would be really useful. If you are on an export tariff where export payments are different throughout the day, it would be great to change these settings quickly in homeassistant

wlcrs, 2022-04-18:
This is already implemented in the underlying library, but it's very hard to expose as configurable entities in HA. There is nothing that has close to all the necessary attributes: start time, end time, day(s) of week + charge/discharge. All of these need to be implemented for every 'line' in FusionSolar

To be able to add this to HA, we need a custom frontend component. If anybody wants to volunteer in making this, I'm open for collaboration!

wlcrs, 2022-04-18:
Tip: you can already look into the "forced discharge" service that you can call in HA to alter the behavior of your battery

jrowland1993, 2022-04-18:
Hi @wlcrs,

Thanks for the reply, appreciate the reasoning. I’ve only just started with home assistant so I lack to knowledge to collab at this stage but hopefully in the not so distant future.

In terms of the forced discharge, I’ve looked at the available field vs the functionality in sun2000 app and they differ a bit. I can’t seem for force discharge from HA. Am I missing something?

View from sun2000:

[image]

inigoml, 2022-04-21:
I also thought about implementing TOU programming but after rethinking, I realize it's much better option to transfer this functionality to home assistant in order to be "more smart".
For example, in my case, instead of using TOU, I force charging at 5AM on very low price period when PV yield prevision is low for next day. It's much better approach than a fixed programming via TOU.
Only problem is (or was) that HA lost connection with inverter and after recovering is not able to send actions (but no problem receiving telemetry), but an easy workaround restarting HA at 4:50AM did the trick.

jrowland1993, 2022-04-22:
@inigoml  - I agree, the force discharge and force charge scripts are much more versatile for home assistant and I've managed to work these into a nice format using a couple of helper fields (input numbers etc) and it works really well for me.

Happy for the Feature request to be closed.

closed 2022-04-22 by jrowland1993