wlcrs/huawei_solar #52: [Bug]: Failed connectivity

JohnMcLear, 2022-04-27:
### System Health details

## System Health

version | core-2022.4.7
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.10.108
arch | aarch64
timezone | Europe/London

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4993
Installed Version | 1.24.0
Stage | running
Available Repositories | 1029
Downloaded Repositories | 8

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.6
-- | --
update_channel | stable
supervisor_version | supervisor-2022.04.0
docker_version | 20.10.9
disk_total | 115.7 GB
disk_used | 92.4 GB
healthy | true
supported | true
board | odroid-n2
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.3.3), MariaDB (2.4.0), Let's Encrypt (4.12.0), Check Home Assistant configuration (3.10.0), Terminal & SSH (9.3.0), Zigbee2mqtt (1.25.0-1), Log Viewer (0.13.0), Mosquitto broker (6.0.1), ESPHome (2022.4.0), Plex Media Server (2.8.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
views | 1
mode | storage

</details>

### Huawei Solar Setup

I'm NOT connecting using the AP method as the AP is 200M away from the HA box.

## Hosts
```
SDongleA-HV2170115623.lan > 10.0.0.101    
SUN2000-102180067169.lan > 10.0.0.182
```

## Firmware
Inverter V200R001C00SPC121
Dongle V100R001C00SPC127

Unrestricted Modbus TCP is enabled and set to RS485.  I did notice that RS485 didn't seem to persist after restarts.

## Connectivity testing
```
nmap SDongleA-HV2170115623.lan -p1-10000
Host is up (0.016s latency).
PORT     STATE SERVICE
502/tcp  open  mbap
6606/tcp open  unknown
```

Note 6606 instead of 6607?!  502 does appear to be open.

### Describe the issue

I'm unable to connect to the inverter or dongle.

I don't seem to have any newer firmware options available in the Fusion web interface.  Should my firmware be okay?
I don't seem to have 6607 available on the Dongle, is that expected?  Do we expect to see 6606?
Is RS485 setting not being persistent a known fault/issue?

### Reproduction steps

1. Install plugin
2. Attempt to connect
3. Fails to connect
...

### Relevant debug logs

```text
Logger: huawei_solar.huawei_solar
Source: /usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py:246
First occurred: 09:22:31 (2 occurrences)
Last logged: 09:23:20

Modbus client is not connected to the inverter.
NoneType: None
```

```
Logger: huawei_solar.huawei_solar
Source: /usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py:270
First occurred: 09:24:21 (2 occurrences)
Last logged: 09:24:28

Could not read register value, has another device interrupted the connection?
```

```
Logger: pymodbus.client.asynchronous.async_io
Source: /usr/local/lib/python3.9/site-packages/pymodbus/client/asynchronous/async_io/__init__.py:307
First occurred: 09:22:30 (48 occurrences)
Last logged: 09:30:09

Failed to connect: [Errno 111] Connect call failed ('10.0.0.101', 6607)
Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 6607, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 6607)
Failed to connect: Multiple exceptions: [Errno 111] Connect call failed ('::1', 6606, 0, 0), [Errno 111] Connect call failed ('127.0.0.1', 6606)
```
```

wlcrs, 2022-04-27:
https://github.com/wlcrs/huawei_solar#what-ip-address-and-port-should-i-enter

closed 2022-04-27 by wlcrs

JohnMcLear, 2022-04-27:
@wlcrs I think something was missed here, 

Please take a moment to read my github issue.

https://github.com/wlcrs/huawei_solar#sdongle-configuration  Applies to my situation.

Note that the docs mention WiFi AP access being the route to take if IP is not available but then the Dongle should be the correct connectivity.

Also note that in my issue I stated I cannot use the WiFI AP method.

Can you clarify, is it the case that one MUST use the WiFi AP method even if a Dongle is present?

Please re-open this issue until we have clarity.

wlcrs, 2022-04-27:
Only the WiFi AP method works on recent firmwares. The presence of a dongle makes no difference anymore.

JohnMcLear, 2022-04-27:
PR sent.  I will setup a tunnel/bridge.  It's super ghetto but I guess that's what you get when you go Huawei ;(

wlcrs, 2022-04-27:
Adding support for a wired connection to the RS485_1 pins is somewhere low on my priority list. This can solve your usecase where it is too hard to connect to the inverter AP. I will need to buy some hardware to connect and test this setup.

JacoboSanchez, 2022-04-27:
FYI I have two inverters with EW11a modbus adapters connected through RS485. One of them is working properly while the other is not very stable and has very long ping times sometimes resulting in "unavailable" states in HA. This is not related to the device as I tried switching them. I am planning on trying another power supply for the RS485 and I am also waiting for a HF2211S model and see If I got better stability results (nothing related with the integration)

JohnMcLear, 2022-04-28:
To wrap this up.  I used a Pi to connect to the AP and used 
```
iptables -t nat -I PREROUTING 1 -p tcp --dport 6607 -j DNAT --to-destination 192.168.200.1:6607
iptables --flush
iptables-save
```

I could then connect.  I wont add this to the docs because I'm not confident it's right, ``iptables`` can be a fickle mistress.