wlcrs/huawei_solar #54: [Bug]:

djwmarcx, 2022-04-29:
### System Health details

## System Health

version | core-2022.4.6
-- | --
installation_type | Home Assistant Container
dev | false
hassio | false
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.4.0-1058-raspi
arch | aarch64
timezone | Europe/Madrid

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.24.5
Stage | waiting
Available Repositories | 1092
Downloaded Repositories | 17

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 10
views | 11
mode | storage

</details>

### Huawei Solar Setup

Inverter: SUN2000-4KTL-L1
Single Phase meter. 
No Battery
No SDongle, builtin wify

### Describe the issue

Both **Grid Active** Power and **Grid Exported** are inverted (negative)

### Reproduction steps

Just connect to my inverter...

### Relevant debug logs

```text
No relevant logs.
```

wlcrs, 2022-04-29:
Please increase the logging level as described in the readme. Otherwise I cannot diagnose the problem

netcraker01, 2022-04-30:
Same problem here ... I make a new template sensor to invert negative value (abs) and use it on energy panel , but it is a temporaly fix... I think that is a issue derivated from energy meter coneccted to Huawei that send values on negative , i´m not sure of this.

wlcrs, 2022-04-30:
I will need extensive logging to verify the values sent by the inverter. Please add the following to your configuration.yaml , restart HA and let it run for a few minutes:

```
logger:
  logs:
    huawei_solar: debug
    homeassistant.components.huawei_solar: debug
```

Other questions:
- what is reported in FusionSolar?
- is the clamp of the power meter correctly attached around the wiring in your electrical cabinet? Verify the direction of the arrow on the clamp. Also verify how the blue/white wires are attached to the power meter.

In my case one clamp was mounted in the wrong direction and another clamp was connected wrong to the power meter. So I don't trust installers to always do the correct thing.

yousaf465, 2022-04-30:
@[djwmarcx](https://github.com/djwmarcx) open fusion app click on your name open statistics from button and scroll down and take a screenshot. Post here. If your green and red curves are separate, Your clamp might be fine

djwmarcx, 2022-04-30:
@wlcrs [wlcrs](https://github.com/wlcrs)
> Please increase the logging level as described in the readme. Otherwise I cannot diagnose the problem

I have updated the original issue with the only logs I have (repeated a bunch of times)

My metering device is a Carlo Gavazzi EM112 (More information [here](https://gavazzi.se/app/uploads/2020/11/em112_im.pdf)). I'm not using clamps.

The Fusion Solar app is working fine since the first day, I've never noticed anything weird (isn't, right? O_O):
[image]

Checking the manual for my meter I confirmed that my numbers are indeed shown on the meter screen as they should (In the correct sign values, not reversed):

[image]

Maybe fusion solar is "fixing" the values itself...

Thanks!!!

netcraker01, 2022-04-30:
Hello I attach a log Im using same meter that @djwmarcx
About questions:
1: On Fusion Solar all is ok , and values are positive
2: I have a Energy meter Carlo Gavazzi EM112,  is a compact modbus RTU Energy meter conected to Inverter, have no isusues with Wires, have a connector to Neutral wire and it is working in a correct way. I attach Datasheet and Modbus registers.

[image]

[Carlo-Gavazzi-EM112-Installation-and-Use-Instructions-EN.pdf](https://github.com/wlcrs/huawei_solar/files/8597326/Carlo-Gavazzi-EM112-Installation-and-Use-Instructions-EN.pdf)
[em111_em112_et112_cp.pdf](https://github.com/wlcrs/huawei_solar/files/8597327/em111_em112_et112_cp.pdf)
[EM112DS.pdf](https://github.com/wlcrs/huawei_solar/files/8597328/EM112DS.pdf)

wlcrs, 2022-04-30:
Grid Active Power being negative at night is consistent with what my inverter reports: this value is positive when I export electricity to the grid during the day, and negative when I draw electricity from the grid at night. Do you also observe this behavior, or do I misunderstand it?

As I understand from the manual your power meter also reports the exported energy with a negative value (the red arrow you draw points to it). This seems to be passed on as-is by the inverter.

The modbus definitions of the power meter are not relevant as this is internal to the inverter, and not exposed to the outside world.

djwmarcx, 2022-05-04:
My Grid Active Power values for last 2 days:
[image]

Yes, my meter is reporting the exported energy as negative.
[image]

I guess the problem is that the inverter is passing the meter information as-is to the Modbus, while the fusion solar is aware of this "negative" value situation, fixing it internally during runtime. This can explain why the Fusion Solar app works fine but the integration is reporting it as negative.

The value we are getting it's correct but negative. Since a negative value won't ever work in the energy dashboard, Can we just apply an ABS function to the value? That will ensure work in both cases.

peaceduck, 2022-05-05:
The thing is, we "need" the active power to be both negative and positive, for other reasons. You do not want to put that into the energy dashboard as-is.
I would recommend that you define 2 template-sensors, and use them. Mine looks like this:
`- name: ”imported electricity”
    unit_of_measurement: "W"
    device_class: power
    state_class: measurement
    state: >-
      {{  -min(0, states('sensor.grid_active_power') | float(0)) }}
  - name: ”Exported ”electricity
    unit_of_measurement: "W"
    device_class: power
    state_class: measurement
    state: >-
      {{ max(0, states('sensor.grid_active_power') | float(0)) }}
`
With those, the energy-dashboard works perfectly!

These should be defined in your block 
`
template:
  sensor:
`

wlcrs, 2022-05-05:
I advice against using `grid_active_power` in the Energy Dashboard as this results in less accurate results: HA will use a Riemann sum to deduce the total energy consumption/export. Use the accumulative values like `grid_exported` and `grid_consumption` instead, as these can be used without conversion (and loss of accuracy) by HA.

More info: https://www.home-assistant.io/docs/energy/faq/#energy-vs-power

peaceduck, 2022-05-05:
My bad... I have given my sensors the same display-name, so I picked the wrong ones when writing here... Of course you should use what @wlcrs writes. Those are also what I have there. They work perfect. The ones I wrote above can be used if you want to display real-time usage of electricity. Sorry for the confusion!

wlcrs, 2022-05-06:
@djwmarcx @netcraker01 Can you please test the fix I implemented in the huawei-solar library version 2.0.2rc1 ?

Change the value in `requirements` from `/config/custom_components/huawei_solar/manifest.json` to:

`huawei-solar==2.0.2rc1`

You might need to restart your HA twice before the download succeeds (due to how PyPI caches it's downloads).

[image]

djwmarcx, 2022-05-06:
it works!

[image]

wlcrs, 2022-05-25:
I've released v1.0.2 which bumps the huawei-solar lib to 2.0.2

closed 2022-05-25 by wlcrs