wlcrs/huawei_solar #57: [Bug]: Permission to write configuration stops after a certain number of hours

peaceduck, 2022-05-04:
### System Health details

## System Health

version | core-2022.4.7
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.10.103-v8
arch | aarch64
timezone | Europe/Stockholm

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4953
Installed Version | 1.24.5
Stage | running
Available Repositories | 1101
Downloaded Repositories | 7

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.6
-- | --
update_channel | stable
supervisor_version | supervisor-2022.05.0
docker_version | 20.10.9
disk_total | 931.4 GB
disk_used | 19.0 GB
healthy | true
supported | true
board | rpi3-64
supervisor_api | ok
version_api | ok
installed_addons | Z-Wave JS (0.1.58), MariaDB (2.4.0), File editor (5.3.3), Let's Encrypt (4.12.2), Samba share (9.6.1), Terminal & SSH (9.4.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 2
views | 2
mode | storage

</details>

### Huawei Solar Setup

Inverter: SUN2000-8KTL-M1 (Software version: V100R001C00SPC141)
Smart Power Meter (3-phase)
Luna 2000 - 5 kWh battery

### Describe the issue

It seems like the permission to write configuration registers stops working after some hours. My guess is that Huawei somehow logs out that permission (or has a timeout).
If I restart home-assistant (or reloads the Huawei integration), the writing works perfect. After a period of time (today 12,5 hours after startup), I get permission denied. It is hard to know exactly when the permission went away. It worked 2,5 hours after startup.
If possible, I would maybe suggest that you do a new login when this happens (or on a timer). I have not looked into that code. Maybe there are some other way of solving it? 

### Reproduction steps

1. Start home assistant
2. Change configuration -> Works ok
3. Wait an unknown number of hours (more than 2.5, less than 12.5)
4. Change configuration -> Permission denied
.

### Relevant debug logs

```text
2022-05-04 07:51:49 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-04 07:51:54 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-04 07:51:56 DEBUG (MainThread) [huawei_solar.huawei_solar] Heartbeat succeeded
2022-05-04 07:51:57 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-04 07:52:04 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-04 07:52:08 DEBUG (MainThread) [huawei_solar.huawei_solar] Writing to 47075: [0, 300]
2022-05-04 07:52:12 DEBUG (MainThread) [huawei_solar.huawei_solar] Heartbeat succeeded

[... cut not relevant messages ...]

2022-05-04 08:23:35 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-04 08:23:35 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-04 08:23:35 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-04 08:23:35 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-04 08:23:48 DEBUG (MainThread) [huawei_solar.huawei_solar] Heartbeat succeeded
2022-05-04 08:24:00 DEBUG (MainThread) [huawei_solar.huawei_solar] Writing to 47075: [0, 250]
2022-05-04 08:24:03 DEBUG (MainThread) [huawei_solar.huawei_solar] Heartbeat succeeded
2022-05-04 08:24:05 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-04 08:24:05 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-04 08:24:05 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-04 08:24:05 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760

[ ... cut ...]

2022-05-04 10:27:05 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-04 10:27:05 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-04 10:27:05 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-04 10:27:16 DEBUG (MainThread) [huawei_solar.huawei_solar] Heartbeat succeeded
2022-05-04 10:27:31 DEBUG (MainThread) [huawei_solar.huawei_solar] Heartbeat succeeded
2022-05-04 10:27:33 DEBUG (MainThread) [huawei_solar.huawei_solar] Writing to 47075: [0, 2500]
2022-05-04 10:27:36 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-04 10:27:36 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-04 10:27:36 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-04 10:27:36 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760

[ ... cut ... ]

2022-05-04 18:29:30 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-04 18:29:30 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-04 18:29:35 DEBUG (MainThread) [huawei_solar.huawei_solar] Writing to 47077: [0, 2400]
2022-05-04 18:29:35 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547167342896] Permission denied
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 193, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1634, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1671, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 204, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 677, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 955, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 714, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 75, in async_set_value
    await entity.async_set_value(value)
  File "/config/custom_components/huawei_solar/number.py", line 197, in async_set_value
    if await self.bridge.set(self.entity_description.key, int(value)):
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/bridge.py", line 265, in set
    return await self.client.set(name, value, slave=self.slave_id)
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 304, in set
    response = await self.write_registers(
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 335, in write_registers
    raise PermissionDenied("Permission denied")
huawei_solar.exceptions.PermissionDenied: Permission denied
2022-05-04 18:30:00 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-04 18:30:00 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-04 18:30:00 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-04 18:30:00 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760

[... Reload Huawei integration ... ]

2022-05-04 18:30:32 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-04 18:30:32 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-04 18:30:36 DEBUG (MainThread) [huawei_solar.huawei_solar] Heartbeat succeeded
2022-05-04 18:30:51 DEBUG (MainThread) [huawei_solar.huawei_solar] Heartbeat succeeded
2022-05-04 18:30:55 DEBUG (MainThread) [huawei_solar.huawei_solar] Writing to 47077: [0, 2499]
2022-05-04 18:31:00 DEBUG (MainThread) [huawei_solar.huawei_solar] Writing to 47077: [0, 2500]
2022-05-04 18:31:02 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-04 18:31:02 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-04 18:31:02 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
```

JacoboSanchez, 2022-05-04:
I think you are describing https://github.com/wlcrs/huawei_solar/issues/28
It does not seems to be a timeout thing but a temporary lost of connection with the inverter which makes the connection to lose the previous authentication

peaceduck, 2022-05-04:
Ok, do you think my router has temporary lost the connection? Because I can not see any problems in Home Assistant. I have not looked in the code, but could it be possible to re-run the authentication code when this happens? Or reconnect to the Huawei? If not possible automatically, maybe publish a service that I could call when this happens?

wlcrs, 2022-05-04:
Can you look into your logs what happens to the heartbeat which keeps the session (with the authentication) open? I'm interested to see which error happens there (if any)

peaceduck, 2022-05-04:
Oh, yes! Sorry, I did not notice that. The heartbeat fails. And never comes back until I reloaded the Integration.
The log:
`2022-05-04 12:52:30 DEBUG (MainThread) [huawei_solar.huawei_solar] Heartbeat succeeded
2022-05-04 12:52:38 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-04 12:52:38 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-04 12:52:38 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-04 12:52:38 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-04 12:52:45 DEBUG (MainThread) [huawei_solar.huawei_solar] Heartbeat succeeded
2022-05-04 12:53:08 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-04 12:53:11 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (asyncio.exceptions.TimeoutError)
2022-05-04 12:53:11 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 2.0 seconds after 1 tries
2022-05-04 12:53:16 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (asyncio.exceptions.TimeoutError)
2022-05-04 12:53:16 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 2.0 seconds after 2 tries
2022-05-04 12:53:19 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-04 12:53:19 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-04 12:53:19 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-04 12:53:49 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-04 12:53:49 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-04 12:53:49 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-04 12:53:49 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-04 12:54:19 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-04 12:54:19 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-04 12:54:19 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-04 12:54:19 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-04 12:54:49 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-04 12:54:49 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-04 12:54:49 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-04 12:54:49 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-04 12:55:19 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-04 12:55:19 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-04 12:55:19 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-04 12:55:19 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-04 12:55:49 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-04 12:55:50 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-04 12:55:50 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-04 12:55:50 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-04 12:56:20 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-04 12:56:20 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-04 12:56:20 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-04 12:56:20 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-04 12:56:50 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-04 12:56:50 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-04 12:56:50 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-04 12:56:50 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-04 12:57:20 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-04 12:57:20 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-04 12:57:20 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-04 12:57:20 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-04 12:57:50 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-04 12:57:50 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-04 12:57:50 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-04 12:57:50 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-04 12:58:20 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-04 12:58:20 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-04 12:58:20 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
`

peaceduck, 2022-05-04:
These errors are then coming back every 5 minutes (approx)
Sometimes with some more error rows:
`2022-05-04 13:12:57 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 2.0 seconds after 1 tries
2022-05-04 13:13:01 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-04 13:13:01 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-04 13:13:01 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-04 13:13:31 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-04 13:13:34 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (asyncio.exceptions.TimeoutError)
2022-05-04 13:13:34 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 2.0 seconds after 1 tries
2022-05-04 13:13:39 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (asyncio.exceptions.TimeoutError)
2022-05-04 13:13:39 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 2.0 seconds after 2 tries
2022-05-04 13:13:44 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (asyncio.exceptions.TimeoutError)
2022-05-04 13:13:44 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 2.0 seconds after 3 tries
2022-05-04 13:13:49 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (asyncio.exceptions.TimeoutError)
2022-05-04 13:13:49 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 2.0 seconds after 4 tries
2022-05-04 13:13:51 ERROR (MainThread) [custom_components.huawei_solar] Timeout fetching BT21A0216315_data_update_coordinator data
2022-05-04 13:14:21 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-04 13:14:24 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (asyncio.exceptions.TimeoutError)
2022-05-04 13:14:24 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 2.0 seconds after 1 tries
2022-05-04 13:14:29 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (asyncio.exceptions.TimeoutError)
2022-05-04 13:14:29 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 2.0 seconds after 2 tries
2022-05-04 13:14:34 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (asyncio.exceptions.TimeoutError)
2022-05-04 13:14:34 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 2.0 seconds after 3 tries
2022-05-04 13:14:39 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (asyncio.exceptions.TimeoutError)
2022-05-04 13:14:39 DEBUG (MainThread) [huawei_solar.huawei_solar] Backing off reading for 2.0 seconds after 4 tries
2022-05-04 13:15:11 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-04 13:15:12 ERROR (MainThread) [huawei_solar.huawei_solar] Could not read register value, has another device interrupted the connection?
2022-05-04 13:15:42 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-04 13:15:42 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-04 13:15:42 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-04 13:15:42 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760`

wlcrs, 2022-05-05:
A small typo prevented the existing re-login logic from working. I've also implemented a secondary retry to make it even more robust: https://gitlab.com/Emilv2/huawei-solar/-/commit/22bd9b4b0e38f69a2206f11e28eeed415cb2644d . I've upgraded the huawei-solar library to 2.0.1 in 1.0.1 of this integration.

closed 2022-05-05 by wlcrs