wlcrs/huawei_solar #58: [Bug]: Still some problems with permission denied

peaceduck, 2022-05-06:
### System Health details

## System Health

version | core-2022.4.7
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.10.103-v8
arch | aarch64
timezone | Europe/Stockholm

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4950
Installed Version | 1.24.5
Stage | running
Available Repositories | 1101
Downloaded Repositories | 7

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.6
-- | --
update_channel | stable
supervisor_version | supervisor-2022.05.0
docker_version | 20.10.9
disk_total | 931.4 GB
disk_used | 19.7 GB
healthy | true
supported | true
board | rpi3-64
supervisor_api | ok
version_api | ok
installed_addons | Z-Wave JS (0.1.58), MariaDB (2.4.0), File editor (5.3.3), Let's Encrypt (4.12.2), Samba share (9.6.1), Terminal & SSH (9.4.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 2
views | 2
mode | storage

</details>

[... 135 lines cut ...]
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1634, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1671, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 204, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 677, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 955, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 714, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 75, in async_set_value
    await entity.async_set_value(value)
  File "/config/custom_components/huawei_solar/number.py", line 197, in async_set_value
    if await self.bridge.set(self.entity_description.key, int(value)):
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/bridge.py", line 275, in set
    return await self.client.set(name, value, slave=self.slave_id)
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 304, in set
    response = await self.write_registers(
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 335, in write_registers
    raise PermissionDenied("Permission denied")
huawei_solar.exceptions.PermissionDenied: Permission denied
```

peaceduck, 2022-05-07:
No problems yet. I am not sure if it has needed to reconnect at all, but for now it looks good!

peaceduck, 2022-05-07:
Oh, but I forgot one thing. The heartbeat disappeared suddenly. Without any errors, warnings or other logs. Don't know if that is a problem?
```
2022-05-06 23:09:40 DEBUG (MainThread) [huawei_solar.huawei_solar] Heartbeat succeeded
2022-05-06 23:09:55 DEBUG (MainThread) [huawei_solar.huawei_solar] Heartbeat succeeded
2022-05-06 23:09:59 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-06 23:09:59 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-06 23:09:59 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-06 23:09:59 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-06 23:10:10 DEBUG (MainThread) [huawei_solar.huawei_solar] Heartbeat succeeded
2022-05-06 23:10:26 DEBUG (MainThread) [huawei_solar.huawei_solar] Heartbeat succeeded
2022-05-06 23:10:29 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-06 23:10:29 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-06 23:10:29 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-06 23:10:29 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-06 23:10:41 DEBUG (MainThread) [huawei_solar.huawei_solar] Heartbeat succeeded
2022-05-06 23:10:56 DEBUG (MainThread) [huawei_solar.huawei_solar] Heartbeat succeeded
2022-05-06 23:10:59 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-06 23:10:59 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-06 23:10:59 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-06 23:10:59 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-06 23:11:11 DEBUG (MainThread) [huawei_solar.huawei_solar] Heartbeat succeeded
2022-05-06 23:11:26 DEBUG (MainThread) [huawei_solar.huawei_solar] Heartbeat succeeded
2022-05-06 23:11:29 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-06 23:11:29 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-06 23:11:29 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-06 23:11:29 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-06 23:11:41 DEBUG (MainThread) [huawei_solar.huawei_solar] Heartbeat succeeded
2022-05-06 23:11:56 DEBUG (MainThread) [huawei_solar.huawei_solar] Heartbeat succeeded
2022-05-06 23:11:59 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-06 23:11:59 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-06 23:11:59 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-06 23:11:59 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-06 23:12:29 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-06 23:12:29 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-06 23:12:29 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-06 23:12:29 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-06 23:12:59 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-06 23:12:59 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-06 23:12:59 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-06 23:12:59 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-06 23:13:29 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-06 23:13:29 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-06 23:13:29 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-06 23:13:29 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760

```

wlcrs, 2022-05-07:
The heartbeat should restart when you do a write-operation. If that's the case, then I can do a proper release of v2.0.2

peaceduck, 2022-05-07:
Sorry, it did not. The write worked, however...
```
2022-05-07 09:56:34 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-07 09:56:34 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-07 09:56:34 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-07 09:56:34 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-07 09:57:04 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-07 09:57:04 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-07 09:57:04 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-07 09:57:04 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-07 09:57:34 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-07 09:57:34 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-07 09:57:34 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-07 09:57:34 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-07 09:57:46 DEBUG (MainThread) [huawei_solar.huawei_solar] Writing to 47075: [0, 1000]
2022-05-07 09:58:04 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-07 09:58:04 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-07 09:58:04 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-07 09:58:04 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-07 09:58:34 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-07 09:58:34 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-07 09:58:34 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-07 09:58:34 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-07 09:59:04 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-07 09:59:04 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-07 09:59:04 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-07 09:59:04 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-07 09:59:34 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-07 09:59:34 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-07 09:59:34 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-07 09:59:34 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-07 10:00:04 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-07 10:00:04 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-07 10:00:04 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-07 10:00:04 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-07 10:00:34 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-07 10:00:34 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-07 10:00:34 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-07 10:00:34 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-07 10:01:04 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-07 10:01:04 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016
2022-05-07 10:01:04 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37101
2022-05-07 10:01:04 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760
2022-05-07 10:01:34 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064
2022-05-07 10:01:34 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32016

```

peaceduck, 2022-05-07:
Lost the heartbeat suddenly again today. Can not see anything happening at that time. Still running debug-log on huawei-things. Everything seems to work fine, though.

peaceduck, 2022-05-12:
I don't know now... I have changed so that it writes registers every hour. (Previously I checked if the state was changed). And now everything have worked perfectly since. So I really think that 2.0.2 is release-worthy.

wlcrs, 2022-05-12:
Thanks for the feedback. I'm currently OoO, so the release will be for somewhere at the end of next week.

cmd85, 2022-05-25:
I have the same issue where I cannot set because of permission denied.
My workaround is that I have in my automations set the integration to "reload" 20s before it tries to write a state which works.

But a proper fix would ofc be best so I can remove my workaround =)

wlcrs, 2022-05-25:
I've released v1.0.2 which bumps the huawei-solar lib to 2.0.2

closed 2022-05-25 by wlcrs