wlcrs/huawei_solar #60: [Bug]: Error setting up entry SUN2000-xxx for huawei_solar

cmd85, 2022-05-25:
### System Health details

## System Health

version | core-2022.4.7
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.10.108
arch | x86_64
timezone | Europe/Stockholm

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4979
Installed Version | 1.24.5
Stage | running
Available Repositories | 1065
Downloaded Repositories | 15

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | June 22, 2022, 02:00
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | eu-west-2-0.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Easee EV Charger</summary>

component_version | 0.9.42
-- | --
reach_easee_cloud | ok
connected2stream | true

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.6
-- | --
update_channel | stable
supervisor_version | supervisor-2022.05.3
docker_version | 20.10.9
disk_total | 62.3 GB
disk_used | 8.7 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Mosquitto broker (6.1.2), File editor (5.3.3), Terminal & SSH (9.4.0), Z-Wave JS (0.1.58), Z-Wave JS to MQTT (0.41.0), Zigbee2mqtt (1.25.1-1)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 9
views | 11
mode | storage

</details>

<details><summary>Nibe Uplink Heat Pumps</summary>

api_endpoint_reachable | ok
-- | --

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Huawei Solar Setup

I am extending the Inverter service wifi to home assistant on a seperate wifi network.
Pinging works fine.

I have Inverter and Battery.

### Describe the issue

**I get this issue where the integration stops working:**
Error setting up entry SUN2000-10KTL-M1SUN2000-12 for huawei_solar
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

**Workaround "fixes":**
To fix this I need to re-dowload via HACS and restart the HA server and it usually jumps on again.

### Reproduction steps

No specifics, let the integratrion run and it will end up here during the day usually.

### Relevant debug logs

```text
Logger: homeassistant.config_entries
Source: custom_components/huawei_solar/__init__.py:67
Integration: Huawei Solar
First occurred: 05:54:33 (1 occurrences)
Last logged: 05:54:33

Error setting up entry SUN2000-10KTL-M1SUN2000-12 for huawei_solar
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 335, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/huawei_solar/__init__.py", line 120, in async_setup_entry
    raise err
  File "/config/custom_components/huawei_solar/__init__.py", line 67, in async_setup_entry
    await primary_bridge.login(
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/bridge.py", line 225, in login
    if not await self.client.login(username, password):
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 374, in login
    login_response = await self._client.protocol.execute(login_request)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/asynchronous/async_io/__init__.py", line 35, in execute
    resp = await asyncio.wait_for(req, timeout=self._timeout)
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
```

wlcrs, 2022-05-25:
I've released v1.0.2 which bumps the underlying library to v2.0.2. This contains the [fix](https://gitlab.com/Emilv2/huawei-solar/-/commit/f0d395291fbc5aa7ccf86d2ce431bb4018eebfda) needed to get the re-authentication workflow to work properly.

Can you please check if this fixes the issue for you?

P.S. Thanks for sponsoring! ❤

sici007, 2022-05-25:
Sorry to hijack the thread, but after trying to update to 1.0.2 from previous release (1.0.1), update failed - error "failed to call service update/install. string index out of range". That happens even if i removed it and want to install 1.0.1 specifically.

cmd85, 2022-05-25:
Thank you, upgraded now and will see.

Could this maybe fix the "permission denied" issue where I have a workaround of reloading the integration also?
Wonder if I should remove that part and see if it works without the workaround.

wlcrs, 2022-05-25:
The periodic reloading of the integration should not be necessary anymore. The re-authentication flow was broken, but has been fixed in v2.0.2 .

closed 2022-05-29 by wlcrs