wlcrs/huawei_solar #61: [Bug]: Unable to update to 1.0.2 - string index out of range

stomko11, 2022-05-25:
### System Health details

Running HA Supervised in VM

### Huawei Solar Setup

Unrelated - integration install issue

### Describe the issue

I am not able to update to 1.0.2 and my integration is gone from integrations. Following error occurs when I want to update, or install 1.0.2 or 1.0.1 which I was running before:

This error originated from a custom integration.

Logger: custom_components.hacs
Source: custom_components/hacs/websocket.py:286
Integration: HACS (documentation, issues)
First occurred: 16:46:32 (2 occurrences)
Last logged: 16:46:40

string index out of range

### Reproduction steps

1. update to 1.0.2

### Relevant debug logs

```text
Logger: custom_components.hacs
Source: custom_components/hacs/websocket.py:286
Integration: HACS (documentation, issues)
First occurred: 16:46:32 (2 occurrences)
Last logged: 16:46:40

string index out of range
```

wlcrs, 2022-05-25:
This is a problem with HACS, not with the integration itself. Does restarting HA help? Is your HACS up-to-date?

I'm not able to reproduce it on my local installation: the upgrade went smoothly.

You can always manually download this repository and put it in your `/config/custom_components`

I'm closing this, as I think you need to contact the HACS team to get this resolved.

closed 2022-05-25 by wlcrs

JacoboSanchez, 2022-05-25:
I am also getting this error... will try removing and downloading again

JacoboSanchez, 2022-05-25:
Not being able to install it with HACS after removing it, for reference the full error is the following:

wlcrs, 2022-05-26:
HI @JacoboSanchez , it seems that the zip-file that is downloaded by HACS is corrupted. However, when I download and test it locally everything is working as expected.

Does the problem persist when you try to install v1.0.1 again? Or is it specific for 1.0.2?

I've already posted this error on the HACS discord, hoping to get some response on there...

stomko11, 2022-05-26:
For me it was the same regardless which version I chose.
After downloading repository manually and copying it to custom components, restarting hA... I was still unable to add it as integration, until I clicked on "add integration" button in install steps here, which navigated me directly to setting it up. Only then I was able to see it.

JacoboSanchez, 2022-05-26:
It did not work to install 1.0.1 nor 1.0.2. I removed the integration, downloaded the repo content manually and added the inverters again and It is working ok. Did not tried to use HACS again yet

2bbe, 2022-05-29:
Same for me…
I get the error while downloading with HACS (1.25.5)
The plug-in itself works with a manual install.

jodewee, 2022-05-30:
same over here, plugin works fine, but you have to download and install it manually