wlcrs/huawei_solar #63: String index out of range

MicheleMercuri, 2022-05-27:
Hi, when I will try to update the integration from 1.01 to 1.02 I get this error
[image]
How I can solve this issue?

MicheleMercuri, 2022-05-27:
Fixed it's a bug of my HA installation. Sorry.

closed 2022-05-27 by MicheleMercuri

wlcrs, 2022-05-27:
Hi, what was the bug in your HA? Other users also experience the same
issue? Can you help them?

MicheleMercuri, 2022-05-27:
I think maybe a lastest hacs update installation problem. But I can't confirm it because I solved by restore a backup made a few days before lastest hacs update (1.25.1) and then update hacs to 1.25.1 again and then update all integration. 
Now all works fine.
First this fix I will try all with no success (unistalled Huawei solar, delete manually, install a previous version 1.0.1).