wlcrs/huawei_solar #64: [Bug]: Timestamps don't match

wlcrs, 2022-05-30:
### System Health details

/

### Huawei Solar Setup

/

### Describe the issue

The times reported by the inverter don't match. This also causes issues when requesting the latest optimizer data, as the timestamps from there are also off by the same amount.

Some debugging info:

On Mon May 30 2022 07:12:53 GMT+0200 (Central European Summer Time), with
```
int(datetime.now().timestamp()) == 1653887573
```
and
```
int(datetime.utcnow().timestamp()) == 1653880373
```

Reading register 40000 resulted in 1653894774.

This is 14400 seconds (=4*60*60) from UTC time, and 7200 seconds (=2*60*60) from the localized timestamp

Meanwhile, my inverter reports the timezone in register `43006` as being 60, matching my (winter timezone) CET.

So it seems that the inverter timestamps move the time into the wrong direction when trying to work with timezones?

### Reproduction steps

1.
2.
3.
...

### Relevant debug logs

```text
/
```

gilder0, 2022-06-11:
According to documentation : https://javierin.com/wp-content/uploads/sites/2/2021/09/Solar-Inverter-Modbus-Interface-Definitions.pdf  

some register eg. Startup Time,  Shutdown time is returned as "Epoch seconds, local time"   but others are returned as "Epoch seconds UTC"

I think the main problem of reading this registeres is here : 
https://gitlab.com/wlcrs/huawei-solar/-/blob/develop/src/huawei_solar/registers.py#L224

"return datetime.fromtimestamp(value - 60 * inverter.time_zone, timezone.utc)"

because it is always interpreted as UTC. 

In my opinion there should by different methods for each registers

gilder0, 2022-06-12:
when i change that line to : 
return datetime.astimezone(datetime.utcfromtimestamp(value), timezone.utc)

now startup and shutdowntime is correctly returned

My timezone is UTC+1  and now during DST on ;)

kulasalu, 2022-07-04:
> when i change that line to : return datetime.astimezone(datetime.utcfromtimestamp(value), timezone.utc)
> 
> now startup and shutdowntime is correctly returned
> 
> My timezone is UTC+1 and now during DST on ;)

How do you change that in a docker based HA environment? I would really like to manually apply the fix if there is no new version coming soon :)

rembomy, 2022-07-04:
> 

How do you change that in a docker based HA environment? I would really like to manually apply the fix if there is no new version coming soon :)

wlcrs, 2022-07-04:
I've applied the suggested change to the library: https://gitlab.com/Emilv2/huawei-solar/-/commit/220b08b3f7891f4ec6cc70a896a8e6a79d3224fb . Thank you @gilder0 !

gilder0, 2024-03-31:
I'm not sure. In which version you applied this fix ?

wlcrs, 2024-03-31:
Something unlucky must have happened with  that commit, as it has disappeared from the history of the library. Possibly due to a force-push operation in the git repository when I was unaware of that consequence.

I have reapplied your commit for the moment, but I now notice that it doesn't work as expected for me.

By chance, during recent development, I added register 32110 (`CURRENT_ELECTRICITY_GENERATION_STATISTICS_TIME`). Last week, when we were still on CET here, the old code gave the expected results for me: it reported the current time when queried. When I checked today, with CEST having gone into effect here, HA interpreted the result as 1 hour into the future.

Applying your fix again made the problem worse: HA interprets the result as 2 hours into the future now.

wlcrs, 2024-03-31:
So a quick summary:

- my inverter reports the timezone in register 43006 as being 60, matching my (winter timezone) CET
- `datetime.fromtimestamp(value - 60 * inverter.time_zone, timezone.utc)` is correct when CET is in effect ...
- but not when CEST has gone into effect.

- The inverter does not expose a register to report if DST is in effect.

I have found no better method to resolve this, then to assume that the client on which this library runs is in the same time zone as the inverter. That way, I can query if DST is in effect according to the local machine

This has resulted into the following code instead: https://gitlab.com/Emilv2/huawei-solar/-/commit/c12585ba7b6cedfb437fcd41df0936754290e1d8 .

gilder0, 2024-03-31:
problem with +2 hour could be 

> Something unlucky must have happened with that commit, as it has disappeared from the history of the library. Possibly due to a force-push operation in the git repository when I was unaware of that [...]

2 hours difference is probably result when you use "fromtimestamp" method.  
Using "utcfromtimestamp" works properly for me without any fuzzy logic with many "if" statement

wlcrs, 2024-03-31:
That method is deprecated in Python 3.12. cfr. https://docs.python.org/3/library/datetime.html#datetime.datetime.utcfromtimestamp

They recommend using `datetime.fromtimestamp(..., timezone.utc)` to create a timezone aware datetime object.

I'm all ears for a simpler solution, but this was the best I could come up with after an hour of tinkering.

gilder0, 2024-04-02:
You are right.

I read more about it and realized that utcfromtimestamp interpret value as local timezone, a then convert it to utc-oriented timestamp.

so i simply try this :
>>> print(datetime.utcfromtimestamp(1712006495).timestamp())
1711999295.0
>>> print(datetime.utcfromtimestamp(1702006495).timestamp())
1702002895.0

which confirms that returned value differs by 2 hours in CEST or 1 hour in CET.

As you mentioned earlier, in python 3.12 utcfromtimestamp method was deprecated, so when using fromtimestamp method values needs to be manually recalculated.

wlcrs, 2024-04-02:
This fix is already live in v1.4.0a1

closed 2024-04-02 by wlcrs

gilder0, 2024-04-06:
I've found simpler solution.
try this :

return datetime.fromtimestamp(value, timezone.utc).replace(tzinfo=None).astimezone(timezone.utc)