wlcrs/huawei_solar #65: [Bug]: Reading of optimizers 'files' support fails

madimadi, 2022-05-30:
### System Health details

Hi,
I'm writing about possible issue with huwaei_solar library.

### Huawei Solar Setup

library:
huawei-solar == 2.1.0a2

Inverter SUN2000-10KTL-M1:
V100R001C00SPC141

Dongle SDongleA:
V100R001C00SPC125

 python --version
Python 3.9.13

### Describe the issue

I managed to have bridge.update() working ok and receiving INVERTER_REGISTERS without issue. I have non zero value in nb_online_optimizers. So function code 0x03 works great.

When trying get optimizer data via `get_latest_optimizer_history_data()` or `get_optimizer_system_information()` I receive in all cases timeouts.

Packet capture shows no response from inverter on 0x41 function code. (Decimal 65)

[image]

After playing through the backoff re-try process it gives up.

[image]

On fusion cloud I can browse my optimizer data including realtime metrics. Any hint on how to troubleshooting this?

### Reproduction steps

Below code is used to run library.
```
import logging
import asyncio

from huawei_solar import HuaweiSolarBridge

FORMAT = ('%(asctime)-15s %(threadName)-15s '
          '%(levelname)-8s %(module)-15s:%(lineno)-8s %(message)s')
logging.basicConfig(format=FORMAT)
_LOGGER = logging.getLogger(__name__)
_LOGGER.setLevel(logging.DEBUG)

async def call_bridge(host):

  primary_bridge = await HuaweiSolarBridge.create(
    host=host,
    slave_id=1,
  )
  from pprint import pprint
  pprint(await primary_bridge.update())
  print("===================================")

  pprint(await primary_bridge.get_optimizer_system_information_data())

_LOGGER.debug("Setup Huawei Inverter")
asyncio.run(call_bridge("192.168.XXX.XXX"))

```

### Relevant debug logs

```text
# python example.py
2022-05-30 13:16:41,555 MainThread      DEBUG    example        :46       Setup Huawei Inverter
{'accumulated_yield_energy': Result(value=303.14, unit='kWh'),
 'active_power': Result(value=1296, unit='W'),
 'daily_yield_energy': Result(value=23.56, unit='kWh'),
 'day_active_power_peak': Result(value=8173, unit='W'),
 'device_status': Result(value='On-grid', unit=None),
 'efficiency': Result(value=96.89, unit='%'),
 'fault_code': Result(value=0, unit=None),
 'grid_frequency': Result(value=49.99, unit='Hz'),
 'input_power': Result(value=1338, unit='W'),
 'insulation_resistance': Result(value=28.84, unit='MOhm'),
 'internal_temperature': Result(value=42.7, unit='°C'),
 'line_voltage_A_B': Result(value=413.5, unit='V'),
 'line_voltage_B_C': Result(value=414.5, unit='V'),
 'line_voltage_C_A': Result(value=410.2, unit='V'),
 'nb_online_optimizers': Result(value=4, unit=None),
 'phase_A_current': Result(value=1.863, unit='A'),
 'phase_A_voltage': Result(value=237.8, unit='V'),
 'phase_B_current': Result(value=1.846, unit='A'),
 'phase_B_voltage': Result(value=238.5, unit='V'),
 'phase_C_current': Result(value=1.852, unit='A'),
 'phase_C_voltage': Result(value=238.6, unit='V'),
 'power_factor': Result(value=1.0, unit=None),
 'pv_01_current': Result(value=1.76, unit='A'),
 'pv_01_voltage': Result(value=279.6, unit='V'),
 'pv_02_current': Result(value=1.74, unit='A'),
 'pv_02_voltage': Result(value=476.7, unit='V'),
 'reactive_power': Result(value=-2, unit='VA'),
 'shutdown_time': Result(value=datetime.datetime(2022, 5, 29, 19, 41, 10, tzinfo=datetime.timezone.utc), unit=None),
 'startup_time': Result(value=datetime.datetime(2022, 5, 30, 4, 10, 59, tzinfo=datetime.timezone.utc), unit=None)}
===================================
2022-05-30 13:16:49,397 MainThread      INFO     _common        :104      Backing off _perform_request(...) for 10.0s (asyncio.exceptions.TimeoutError)
2022-05-30 13:17:02,415 MainThread      INFO     _common        :104      Backing off _perform_request(...) for 10.0s (asyncio.exceptions.TimeoutError)
2022-05-30 13:17:15,430 MainThread      INFO     _common        :104      Backing off _perform_request(...) for 10.0s (asyncio.exceptions.TimeoutError)
2022-05-30 13:17:28,448 MainThread      INFO     _common        :104      Backing off _perform_request(...) for 10.0s (asyncio.exceptions.TimeoutError)
2022-05-30 13:17:41,455 MainThread      INFO     _common        :104      Backing off _perform_request(...) for 10.0s (asyncio.exceptions.TimeoutError)
2022-05-30 13:17:54,472 MainThread      ERROR    _common        :119      Giving up _perform_request(...) after 6 tries (asyncio.exceptions.TimeoutError)
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 490, in wait_for
    return fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/backoff/_async.py", line 133, in retry
    ret = await target(*args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 315, in _perform_request
    response = await self._client.protocol.execute(request)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/asynchronous/async_io/__init__.py", line 35, in execute
    resp = await asyncio.wait_for(req, timeout=self._timeout)
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/app/example.py", line 47, in <module>
    asyncio.run(call_bridge("192.168.XXX.XXX"))
  File "/usr/local/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 647, in run_until_complete
    return future.result()
  File "/app/example.py", line 44, in call_bridge
    pprint(await primary_bridge.get_optimizer_system_information_data())
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/bridge.py", line 233, in get_optimizer_system_information_data
    file_data = await self.client.get_file(
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 370, in get_file
    result = await _do_read_file()
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 330, in _do_read_file
    start_upload_response = await _perform_request(
  File "/usr/local/lib/python3.9/site-packages/backoff/_async.py", line 141, in retry
    await _call_handlers(on_giveup, *details)
  File "/usr/local/lib/python3.9/site-packages/backoff/_async.py", line 34, in _call_handlers
    await hdlr(details)
  File "/usr/local/lib/python3.9/site-packages/backoff/_async.py", line 16, in f
    return coro_or_func(*args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 297, in backoff_giveup
    raise ReadException(
huawei_solar.exceptions.ReadException: Failed to read file 69 after 6 tries
```
```

wlcrs, 2022-05-30:
I would expect your code to fail with a permission error. because you didn't login.

I've tested the 'files'-support with the following:

```
async def test():

    bridge = await HuaweiSolarBridge.create("192.168.200.1", port=6607)
    await bridge.login("installer", "00000a")
    await asyncio.sleep(5)

    print(await bridge.get_latest_optimizer_history_data())
```

Maybe you're also going a bit too fast for the inverter? Does it help when you explicitly add the sleep after connecting+login?

wlcrs, 2022-05-30:
Before the update to the firmware that closed the 502 port of the inverter on my home network, I could successfully perform the login via that host/port. So I expect it to just work in your case.

madimadi, 2022-05-30:
Tried login as you suggested (despite the fact it reads registers successfully without it) and doing sleep. 

```
Traceback (most recent call last):
  File "/app/example.py", line 51, in <module>
    asyncio.run(call_bridge("192.168.0.179"))
  File "/usr/local/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 647, in run_until_complete
    return future.result()
  File "/app/example.py", line 38, in call_bridge
    await primary_bridge.login("installer", "00000a")
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/bridge.py", line 284, in login
    if not await self.client.login(username, password):
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 441, in login
    assert challenge_response.content[0] == 0x11
AttributeError: 'ExceptionResponse' object has no attribute 'content'
```
About the optimizer realtime data, do you know if 
```
Inverter SUN2000-10KTL-M1:
V100R001C00SPC141

Dongle SDongleA:
V100R001C00SPC125
```
supported?

wlcrs, 2022-05-30:
That error shows that the first login-command returns a Modbus exception instead of the expected response. Your inverter doesn't seem to like "private" (as in Huawei-specific) modbus requests on its public interface on port 502. Both the login and "file upload" use the private 0x41 Modbus command.  

I would suggest that you setup a WiFi bridge with the `SUN2000-XXXX` network if you want to continue testing. This is also the interface on which the FusionSolar "device commissioning" app works, which I had to partially reverse-engineer to understand the login-sequence and files-support. If the FusionSolar "device commissioning"  app can connect to your inverter, then I would expect the integration to work as expected too.

I cannot confirm/deny the support of any inverter/firmware combination, except for my own SUN2000-3KTL-L1 which is running the latest? SPC119 firmware for the moment. I do not have access to other setups to test my code.

madimadi, 2022-05-31:
You are right, I confirm by connecting to wifi signal of the inverter `SUN2000-XXX` and using `HuaweiSolarBridge.create("192.168.200.1", port=6607)`  as you said above all registers and `get_latest_optimizer_history_data()` and `get_optimizer_system_information()`  method succeeds.
Let me summarize my finding:

Connection to inverter via home lan network TCP/502
- no auth required
- can read all INVERTER_REGISTERS
- cannot access optimizer data (no response to 0x41 Modbus command, not even an error msg)
- slave_id must be set (eg: slave_id=1 )

Connection to inverter via its Wifi AP (SSID: SUN2000-XXXX) TCP/6607:
- auth required
- can read all INVERTER_REGISTERS
- can access optimizer data (via calling  `get_latest_optimizer_history_data()` and `get_optimizer_system_information()`  methods)
- slave_id  not specified at create()

closed 2022-05-31 by madimadi