wlcrs/huawei_solar #66: [Bug]: [139760064938432] Permission denied When setting values

cmd85, 2022-05-31:
### System Health details

## System Health

version | core-2022.5.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.15.41
arch | x86_64
timezone | Europe/Stockholm

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4905
Installed Version | 1.25.5
Stage | running
Available Repositories | 1046
Downloaded Repositories | 15

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | June 22, 2022, 02:00
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | eu-west-2-0.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Easee EV Charger</summary>

component_version | 0.9.43
-- | --
reach_easee_cloud | ok
connected2stream | true

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 8.1
-- | --
update_channel | stable
supervisor_version | supervisor-2022.05.3
docker_version | 20.10.14
disk_total | 62.3 GB
disk_used | 10.9 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Mosquitto broker (6.1.2), File editor (5.3.3), Terminal & SSH (9.4.0), Z-Wave JS (0.1.60), Z-Wave JS to MQTT (0.42.0), Zigbee2mqtt (1.25.1-2), Studio Code Server (5.0.4)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 9
views | 11
mode | storage

</details>

<details><summary>Nibe Uplink Heat Pumps</summary>

api_endpoint_reachable | ok
-- | --

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Huawei Solar Setup

Sun2000 Inverter and Sun2000 battery 

### Describe the issue

I noticed that my automations stopped working and notice that I get "[139760064938432] Permission denied" when changing a value via automations or in the Device page.

Seems to work if I try and change it twice.
I had some similiar things before where I reloaded the integration before any change, is that something I should try again?

### Reproduction steps

Let the system run for a while and try change battery settings via automation or device page

### Relevant debug logs

```text
LOG CHANGE 1:
--------------------------------
Logger: huawei_solar.bridge
Source: /usr/local/lib/python3.9/site-packages/huawei_solar/bridge.py:339
First occurred: 01:00:59 (2 occurrences)
Last logged: 08:42:03

Could not login to set storage_charge_from_grid_function .
Could not login to set storage_working_mode_settings .

--------------------------------
LOG CHANGE 2:
--------------------------------
Logger: homeassistant.components.websocket_api.http.connection
Source: custom_components/huawei_solar/select.py:166
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 08:42:03 (1 occurrences)
Last logged: 08:42:03

[139760064938432] Permission denied
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 189, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1627, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1664, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 204, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 677, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 964, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 714, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 54, in async_select_option
    await entity.async_select_option(option)
  File "/config/custom_components/huawei_solar/select.py", line 166, in async_select_option
    await self.bridge.set(
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/bridge.py", line 340, in set
    raise err
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/bridge.py", line 333, in set
    return await self.client.set(name, value, slave=self.slave_id)
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 395, in set
    response = await self.write_registers(
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 426, in write_registers
    raise PermissionDenied("Permission denied")
huawei_solar.exceptions.PermissionDenied: Permission denied

--------------------------------
LOG CHANGE 3:
--------------------------------

This error originated from a custom integration.

Logger: custom_components.huawei_solar
Source: helpers/update_coordinator.py:197
Integration: Huawei Solar
First occurred: May 30, 2022, 22:35:00 (5 occurrences)
Last logged: 08:11:31

Timeout fetching BT21A0134715_data_update_coordinator data
```

wlcrs, 2022-05-31:
Thank you for the informative bug report. I had enough information to find [the bug in the re-login flow](https://gitlab.com/Emilv2/huawei-solar/-/commit/e814beb75ff6686fd8e87fba63934733dd54dba3).

You can already update your integration by setting [the requirements in `manifest.json`](https://github.com/wlcrs/huawei_solar/blob/develop/manifest.json#L7) to `huawei-solar==2.1.0a4`. I'll do a proper (beta) release once I get feedback on [another issue](https://gitlab.com/Emilv2/huawei-solar/-/issues/14) that I want to resolve.

closed 2022-05-31 by wlcrs

cmd85, 2022-05-31:
Thank you, will update and report if I see any issues!