wlcrs/huawei_solar #67: [Feature Request]: Excess power destination

javierin, 2022-06-02:
### Describe your feature request

When using a battery together with the SUN2000 you can configure where you want to prioritize excess PV output. You can either use it to charge the battery until 100% or always feed the grid with excess power.

Would it be possible to implement this? 
reg 47299 set to 0 means Feed excess PV energy to grid
reg 47299 set to 1 means Charge the battery with excess PV energy

Thanks

wlcrs, 2022-06-03:
Good feature request. However, I will only have time to implement it somewhere in July. PR's are welcome in the meantime.

wlcrs, 2022-10-30:
This request fell of my radar. It will be included in v1.1.5 that I'll release shortly.

closed 2022-10-30 by wlcrs