wlcrs/huawei_solar #68: [Bug]: Ethernet connection documentation

luzik, 2022-06-04:
### System Health details

xxx

### Huawei Solar Setup

HA - LANa - OpenWRT - LANb - SDongle (Ethernet cable)

### Describe the issue

Documentation in multiple places of wiki and README states that connection via dongle _does not work anymore in firmware versions starting from December 2021_

I have working connection over Ethernet SDongle. My inverter firmware is ...SPC145 and SDongle version ...SPC127

I am using this with https://github.com/Emilv2/huawei_solar 

### Reproduction steps

1.
2.
3.
...

### Relevant debug logs

```text
$ telnet 192.168.124.13 502                                                                           
Trying 192.168.124.13...
Connected to 192.168.124.13.
Escape character is '^]'.
Connection closed by foreign host.
```
```

wlcrs, 2022-06-04:
Thank you for the feedback. Can you add your config to https://github.com/wlcrs/huawei_solar/wiki/Confirmed-firmware ?

I need more community input before I change the wording. Otherwise the number of support requests about this will rise 😕

closed 2022-06-04 by wlcrs

luzik, 2022-06-09:
I installed v1.0.2 and it do work over FastEthernet.
So I updated wiki.
It worked fine with version 123 of dongle (:502 unit=1). Version 124 was requires inverter firmware upgrade and also new phone app with new tcp modbus options turned on (that you describe in docs). 124 version just turned tcpmodbus off by default, but allows to turn it on via fusion mobile app

luzik, 2022-06-09:
I belieave that I read it in firmware changelog (by Huawei) somewhere - it was in PDF