wlcrs/huawei_solar #70: [Feature Request]: house consumption/load

tzapu, 2022-06-08:
### Describe your feature request

Hello and thank you for this amazing addon. 
This must be a really silly thing to ask but i can t figure it out for the life of me, how could I get a sensor with the whole shows consumption in kwh ... just a counter that increases or something ,that ignore what s being sent to the grid, etc... need to keep track of my consumers as well

cheers

MiguelAngelLV, 2022-06-09:
You can create a template sensor:

``ỳaml

template:
    - name: House consumption
      unit_of_measurement: W
      state: "{{ max(states('sensor.active_power') | float - states('sensor.grid_active_power') | float, 0) }}"

```

tzapu, 2022-06-09:
thank you very much for the pointer, will try to sort it similar for energy as well 🙌