wlcrs/huawei_solar #73: [Feature Request]: add "Charge To Grid" working mode

VederJay, 2022-06-10:
### Describe your feature request

There's already the option in this integration to set the working mode to either "Maximise Self Consumption" or "Time Of Use". The third working mode is "Charge To Grid". I can set this third option through the fusionsolar app (when logged in as installer), but not through this integration. 
I'm hoping this would be a minor change as the config entry already exists, maybe it's just an enum to be added or some such?

[image]

VederJay, 2022-06-10:
Works perfectly. Thanks for the speedy response and happy holidays. 
I'll leave the ticket open so there's a reminder for adding it to a release at some point :)

wlcrs, 2022-07-28:
This was released in 1.1.1

closed 2022-07-28 by wlcrs