wlcrs/huawei_solar #74: Battery temperatura

epserra, 2022-06-11:
### Describe your feature request

can you include the battery temperature register 3702 

[image]

wlcrs, 2022-07-04:
The SUN2000 inverters support two batteries, as you already saw in the documentation by the duplication of [Energy storage unit 1] and [Energy storage unit 2] registers: `37xxx` and `38xxx` . I made the choice of only exposing the values that the inverter reports where it combines the info off both batteries: `47xxx`.

Adding the temperature battery would mean that I need to rethink this whole approach, as it is possible for people to have two batteries with different temperatures. 

I'm leaving this issue open to see if there are other people that really want this feature, because I currently do not have the time available to do this work.

epserra, 2022-07-05:
Thanks for the info.
I have reviewed the information and register 37022 refers to the temperature of module 1 of batteries, 37752 indicates the temperature of module 2 (I think it is the control of the 3 batteries that each module supports).
Registers 38452 to 38463 indicate the maximum and minimum temperature of each battery.
I believe that registers (No. 12) 37022 storage control temperature 1 and (No.29) 37752 storage control temperature 2, are the important registers to be able to cool the battery block.
If you consider it valid, you can keep it in mind and incorporate it in future improvements.

tarvotarvo, 2022-11-22:
Hi,
and yes, it would be really helpful to get the battery temperature straight from the internal sensor. Currently I have a separate sensor dropped between the cooling fins and its value is used to activate an additional cooling fan when the battery gets really hot.
Please incorporate the temperature register(s) when you find some free time. 
Many thanks in advance!

closed 2023-01-29 by stale[bot]

spelzl, 2024-02-05:
I am interested to get the individual battery sensor data added to the HA integration.
Background is the work of this project:

https://github.com/dalathegreat/Battery-Emulator/issues/15

Roving-Ronin, 2024-02-09:
+2

ebcjaa, 2024-02-16:
I would say that this is quite useful as battery capacity is degraded at low temperatures