wlcrs/huawei_solar #77: [Bug]: 'Efficiency' reports wrong percentage

jov58, 2022-06-13:
### System Health details

-

### Huawei Solar Setup

I use this nice integration for my SUN2000-12KTL-M2 inverter with a SmartDongle and UTP cable.

### Describe the issue

The field `Efficiency` reports the value `655.35%` (FFFF?) when the inverter is stand-by and obviously doesn't represent the real value. I suggest to replace this value by `0.0%` (or `unvailable`?) so graphs and other data views are neater.

### Reproduction steps

1. Install the integration.
2. Look at the sensor `Efficiency `when the inverter is stand-by.

### Relevant debug logs

```text
-
```

wlcrs, 2022-06-27:
Can you please let me know which version of the integration you are using?

wlcrs, 2022-07-04:
Closed due to inactivity

closed 2022-07-04 by wlcrs

luzik, 2022-07-26:
I am using the newest one, and issue is still there

wlcrs, 2022-07-26:
Revisited the code in question, and found a lingering bug: https://gitlab.com/Emilv2/huawei-solar/-/commit/e9a3dcd521bd93d1ba762a0534ea2e706fdfd19b