wlcrs/huawei_solar #79: [Feature Request]: Operating status/ON-OFF switch

geraldFS, 2022-06-20:
### Describe your feature request

Hello!

Is it possible to get the status from the inverter?

- Standby
- Operation
- out

and possibly restart the inverter?

Wouldn't be so wrong sometimes, at the weekend there was a case that the inverter switched itself off due to a thunderstorm.
Undervoltage in one phase was detected.

The restart was only possible via a switch on the device or via the web interface on the Huawei.

Thanks!

wlcrs, 2022-06-27:
The inverter exposes a lot of state, which cannot be easily exposed in home assistant into one entity, as multiple states can be active at once:

[image]
[image]

Note that there is no 'out' status, but it does have a lot of alarm bit-codes. I could expose the active alarms as a comma-separated list of text. This would allow you to check for values via jinja-filters in HA.

I'm not aware of a restart-functionality via Modbus. You'll need to do this manually/via FusionSolar web

tomderidder, 2022-08-29:
@wlcrs Is it feasable to build in some code to expose alarms etc.? We have quite some alarms from the inverter lately due to high voltage, and it would be great  to have this information in ha as well...

(And thanks a lot for the great job you did!)

numanx11, 2022-10-09:
@wlcrs It would be great to add just the Shutdown feature or reset.
Sometimes it happens that the inverter connectivity to Huawei NMS fails and a reboot/reset for the inverter helps.

Meanwhile the inverter is accessible from the local network.

wlcrs, 2022-10-29:
The extra state and alarm entities have been added in [v1.1.4](https://github.com/wlcrs/huawei_solar/releases/tag/1.1.4). 

As there is no documented way to perform a restart of the inverter via Modbus, I cannot add this feature.

closed 2022-10-29 by wlcrs