wlcrs/huawei_solar #81: [Warning]: 2 warnings when using the new core-2022.7.0b0

gieljnssns, 2022-06-29:
### System Health details

## System Health

version | core-2022.7.0b0
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.15.45
arch | x86_64
timezone | Europe/Brussels

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4497
Installed Version | 1.25.5
Stage | running
Available Repositories | 1145
Downloaded Repositories | 38

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 8.2
-- | --
update_channel | beta
supervisor_version | supervisor-2022.06.2
agent_version | 1.2.1
docker_version | 20.10.14
disk_total | 457.7 GB
disk_used | 118.3 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Samba share (9.7.0), AdGuard Home (4.6.0), AppDaemon (0.9.0), Caddy (1.6), Check Home Assistant configuration (3.11.0), Duck DNS (1.15.0), ESPHome (2022.6.2), Mosquitto broker (6.1.2), Studio Code Server (5.1.2), WireGuard (0.7.0), MariaDB (2.4.0), SSH & Web Terminal (11.0.1), Glances (0.16.0), phpMyAdmin (0.8.0), Mealie (1.0.0.1), Arpspoof (1.0.0), Caddy 2 (1.2.0), RTSPtoWeb - WebRTC (1.2.2), EMHASS (0.2.19), Frigate NVR (Full Access) (3.1), Home Assistant Google Drive Backup (0.108.2), Syncthing (1.14.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 18
views | 18
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 23 juni 2022 21:14
-- | --
current_recorder_run | 30 juni 2022 00:53
estimated_db_size | 4208.86 MiB
database_engine | mysql
database_version | 10.4.19

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Huawei Solar Setup

SUN2000-4KTL-M1SUN2000-5KTL-M

### Describe the issue

There are 2 deprecation warnings

### Reproduction steps

1. Update to beta
2. Restart HA
3.
...

### Relevant debug logs

```text
Logger: homeassistant.components.number
Source: components/number/__init__.py:152
Integration: Nummer (documentation, issues)
First occurred: 00:54:10 (6 occurrences)
Last logged: 00:54:10

custom_components.huawei_solar.number is setting deprecated attributes on an instance of NumberEntityDescription, this is not valid and will be unsupported from Home Assistant 2022.10. Please report it to the custom component author.

Logger: homeassistant.components.number
Source: components/number/__init__.py:220
Integration: Nummer (documentation, issues)
First occurred: 00:54:10 (1 occurrences)
Last logged: 00:54:10

custom_components.huawei_solar.number::HuaweiSolarNumberEntity is overriding deprecated methods on an instance of NumberEntity, this is not valid and will be unsupported from Home Assistant 2022.10. Please report it to the custom component author.
```

wlcrs, 2022-07-04:
Fixing this warning will make the integration fail on 2022.6 and earlier. I'll make the necessary changes somewhere in September to allow for a smooth transition.

wlcrs, 2022-07-28:
This has been fixed since 1.1.1

closed 2022-07-28 by wlcrs