wlcrs/huawei_solar #84: [Bug]: grid consumption history on the energy dashboard is showing an abnormal aggregate

ripcam21, 2022-07-08:
### System Health details

[image]

### Huawei Solar Setup

Inverter - SUN20006KTL-l1
no sdongle
three phase power
Battery yes - Huawei luna 10KW
HASSIO- 8.2 (2022.07.0)

### Describe the issue

When using the default energy dashboard, everything is working correctly however when you toggle between weeks, months, or years, the grid consumption aggregate is abnormally high and makes the data insights useless.  Can this be fixed?

[image]
[image]

### Reproduction steps

1.go to energy dashboard
2.select the weeks card
3.toggle between dates
...

### Relevant debug logs

```text
No debug logs to report
```

wlcrs, 2022-07-08:
What type of power meter do you have? Is it the DTSU666-H or another one?

[image]

Which version of the integration are you running?

In recent versions of the integrations, I've fixed a few issues that could have caused "invalid" readings to be passed on as huge numbers to HA. Can you check when that reading took place for you: in which month did this false value pop up? 

You should be able to manually fix this error in HA via 'Developer tools' -> 'Statistics'.

ripcam21, 2022-07-08:
I'm using the latest version of the integration, and the sensor I have is the same as the above. DTSU666 - H 250A/50mA.

I have looked at the statistics table and can see two dates recently which lead to a sudden increase in the sum column, and some instances in the state column.  One occurred on the 2/jul/22, the other 26/jun/2022.  The issue I noticed is that once the data is reported once incorrectly in the statistics table, it forever continues to aggregate off that data.  So day data will show correctly the next day, however when viewing by month, week, or year, you're stuck with the incorrect aggregate.

I did try to update the table with the correct values, however, on the next hour update, the sum column again showed the increased aggregate.  I cannot figure out why that is.

wlcrs, 2022-07-11:
Your csv indeed shows a jump in the `battery_total_charge`, but this is not due to a erroneous reading from the integration: the inverter is sending these values... 

Can you see a pattern when these jumps happen? For example when the battery stops discharging due to the minimum DoD being reached?

Can you also clarify how the 'grid consumption' and 'grid exported' are defined in your installation? This integration does not release "ready to be used" sensors for this.

wlcrs, 2022-07-28:
I have a feeling that this issue will be fixed in 1.1.2, as per https://gitlab.com/Emilv2/huawei-solar/-/commit/e9a3dcd521bd93d1ba762a0534ea2e706fdfd19b . Can you please let me know if you see these jumps in your data again? Then I'll reopen this issue.

Side note: I think that the root cause for these errors might be a cabling problem between the inverter and the power meter, causing the readings to fail from time to time. It would not hurt to check if everything is well connected.

closed 2022-07-28 by wlcrs