wlcrs/huawei_solar #88: [Bug]:  Unable connect to Huawei SUN2000 30-KTL-M3

bdkacz, 2022-07-16:
### System Health details

## System Health

version | core-2022.7.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.15.45
arch | x86_64
timezone | Europe/Warsaw
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4906
Installed Version | 1.26.2
Stage | running
Available Repositories | 1071
Downloaded Repositories | 3

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 8.2
-- | --
update_channel | stable
supervisor_version | supervisor-2022.07.0
agent_version | 1.2.1
docker_version | 20.10.14
disk_total | 30.8 GB
disk_used | 6.3 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.3.3), Let's Encrypt (4.12.5), Samba share (10.0.0), Log Viewer (0.14.0), ESPHome (2022.6.2), SSH & Web Terminal (11.0.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
views | 2
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 6 lipca 2022 11:52
-- | --
current_recorder_run | 16 lipca 2022 08:41
estimated_db_size | 80.94 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

<details><summary>Sonoff</summary>

version | 3.1.0 (3871c0b)
-- | --
cloud_online | 2 / 4
local_online | 0 / 0

</details>

### Huawei Solar Setup

Router with dd-wrt as clienjt is connected to wi-fi of inverter and I make static route on my  main router. HA Can ping 192.168.8.1 (IP of inverter)

### Describe the issue

When I try add integration  with parameters

Host: 192.168.8.1
Port 502
Slave ID :0 

I get error "Reading from the inverter failed."

Emilv2 integration works - so the connection itself  seems working ok

### Reproduction steps

1. Install integration and restart HA
2. Try add inverter as described in beggining of "Describe the issue"
3.
...

### Relevant debug logs

```text
2022-07-16 08:46:02 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting to 192.168.8.1:502.
2022-07-16 08:46:02 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Connecting.
2022-07-16 08:46:02 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client connected to modbus server
2022-07-16 08:46:02 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol made connection.
2022-07-16 08:46:02 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Connected to 192.168.8.1:502.
2022-07-16 08:46:03 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 43006
2022-07-16 08:46:03 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0xa7 0xfe 0x0 0x1
2022-07-16 08:46:03 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 1
2022-07-16 08:46:03 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x0 0x1 0x0 0x0 0x0 0x5 0x0 0x3 0x2 0x0 0x3c
2022-07-16 08:46:03 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x1 0x0 0x0 0x0 0x5 0x0 0x3 0x2 0x0 0x3c
2022-07-16 08:46:03 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse: 3]
2022-07-16 08:46:03 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 1
2022-07-16 08:46:03 DEBUG (MainThread) [pymodbus.payload] [60]
2022-07-16 08:46:03 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47000
2022-07-16 08:46:03 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0x2 0x0 0x0 0x0 0x6 0x0 0x3 0xb7 0x98 0x0 0x1
2022-07-16 08:46:03 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 2
2022-07-16 08:46:03 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x0 0x2 0x0 0x0 0x0 0x3 0x0 0x83 0x2
2022-07-16 08:46:03 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x2 0x0 0x0 0x0 0x3 0x0 0x83 0x2
2022-07-16 08:46:03 DEBUG (MainThread) [pymodbus.factory] Factory Response[131]
2022-07-16 08:46:03 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 2
2022-07-16 08:46:03 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] Client disconnected from modbus server: None
2022-07-16 08:46:03 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol lost connection.
```

bdkacz, 2022-07-16:
I downloaded SUN2000MA V100R001C20SPC116 Release Documents  -   documents  for my firmware version of inverter and go to Solar Inverter Modbus Interface Definitions (V3.0).pdf and I don't see  register 47000 ih documentation . There is few started with 470  releated with Energy storage but nothing with 00 at end - maybe thismis the problem ?

wlcrs, 2022-07-16:
Can you please send me that pdf? Dragging it into the reply dialog here should allow you to attach it to your reply. Does your inverter have support for batteries?

bdkacz, 2022-07-16:
> Can you please send me that pdf? Dragging it into the reply dialog here should allow you to attach it to your reply. Does your inverter have support for batteries?

Link to documentation  [click](https://drive.google.com/file/d/1rvwU4Sjhp2AuEAG5sBOsZ-FK2NIn_Edq/view?usp=sharing)

As for battery support, I do not know - I was not interested in it

bdkacz, 2022-07-16:
This is commercial model so it is posisble that it have different registry<>value map :)

wlcrs, 2022-07-16:
The automatic detection for battery support normally catches ReadException's due to that register not existing. I'll need to double check what is going on.

Feel free to have a look at the code yourself (https://gitlab.com/Emilv2/huawei-solar/-/tree/develop/ ), as fixing bugs without having the hardware present is very hard.

bdkacz, 2022-07-16:
It is strange that both integrations for communication use pymodbus, so if the connection works on one integration, then on the other one based on the same should also...

wlcrs, 2022-07-18:
Can you please try again with version 1.1.0?

bdkacz, 2022-07-18:
> Can you please try again with version 1.1.0?

Inverter added successfully and shows values so it's probably ok :)

BTW, can you write what was the reason that earlier versions did not work?

wlcrs, 2022-07-18:
The following commit fixed the issue: https://gitlab.com/Emilv2/huawei-solar/-/commit/daa2f66f1bb7b8ded4a866a015d48ad5d247d07f

closed 2022-07-18 by wlcrs

bdkacz, 2022-07-18:
One more question -  on energy tab I use sensor of total production  from emilv2 integration - is it easy way to replace it by your sensor without history data lost ?